fivdi/onoff #112: Almost all relays stay on after adding a Darlington transistor array between gpio and relay IN

bmwertman, 2018-06-07:
I started my project by wiring GPIOoutput configured as; `const relay = new Gpio(17, 'out')` for example. I was able to toggle my relays in my node program on and off as commanded. 

When I expanded the program to include more relays I noticed I was getting relays changing from open to closed or vise versa that were not being commanded at the same time as commanded relays were actuated. This lead me to seek guidance on whether I had mis-wired something on the Raspberry Pi forum and Stack Exchange's Raspberry Pi Q&A site. 

The advice that I received on the Raspberry Pi forum seemed like a reasonable solution, add a Darlington transistor between the GPIO out signal and the relay IN. While doing this I added 3 additional relays that I knew would be needed for my project while soldering the Darlington to my Pi hat.

What I found after adding this was that all of the originally configured relays stay on from the time my Pi boots up and that I am unable to control any of them. I see the relay's LED go between bright and dim when I send the open and closed command to them but no relay click to indicate they are actually opening/closing. I have not tried to add the GPIO channels of the three new relays to my program yet and they do not come on when the pi boots. 

After reading other issues such as [Switch on and off a lamp using relay  #102](https://github.com/fivdi/onoff/issues/102) and [All relays turn on when starting the app  #57](https://github.com/fivdi/onoff/issues/57) I tried reconfiguring my existing GPIO channels to use either `'high'` or `'low'` in their configuration. After powering down, unplugging and rebooting my Pi the GPIO that were staying actuated(LED light on) continued to do so.

Here is a picture/wiring diagram of my hat and relays 
[image]
 
Relays 1, 2 and 3 are the ones that are new and do not have their LED's light(ie come on or actuate). Relays 4, 5, 6 and 7 are the original relays to the project that have their LED's light up and they actuate when my Pi boots up. For reference here are links to the threads of my Raspberry Pi forum post and the Stack Exchange Raspberry Pi Q&A post.

https://raspberrypi.stackexchange.com/questions/84410/gpio-relay-signal-cross-turns-to-all-relays-staying-on-after-darlington-added

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=214622

fivdi, 2018-06-08:
I'm afraid I can't provide any additional information that hasn't already be provided in the StackExchange or Raspberry Pi forum posts linked to above. It's almost certainly a hardware issue rather than a software issue.

bmwertman, 2018-06-08:
Thanks. I just thought it seemed like a good possibility that it might be a hardware issue because the three unconfigured relays were not showing the same behavior as those that were previously configured did.

bmwertman, 2018-06-12:
OK, I completed updated my wiring and replaced a hat and got all my relays working as verified by this Python program;
```
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
pins = [5, 16, 22, 24, 25, 26, 27]

for pin in pins :
        GPIO.setup(pin, GPIO.OUT)
        a = 1
        while a < 5 :
                GPIO.output(pin, GPIO.HIGH)
                print pin , " high"
                time.sleep(3)
                GPIO.output(pin, GPIO.LOW)
                print pin , " low"
                time.sleep(3)
                a = a + 1       
        print "finished"
```
But...

When I run run this program;
```
const raspi = require('raspi'); // https://github.com/nebrius/raspi
const Gpio = require('onoff').Gpio; // https://github.com/fivdi/onoff+
const LocalStorage = require('node-localstorage').LocalStorage; // https://github.com/lmaccherone/node-localstorage
const schedule = require('node-schedule'); // https://github.com/node-schedule/node-schedule
const empty_sump_float = new Gpio(4, 'in', 'rising');
const full_sump_float = new Gpio(6, 'in', 'rising');
const drain_relay = new Gpio(22, 'high');
const fill_relay = new Gpio(27, 'high');
const feed_relay = new Gpio(25, 'high');
const rodiValve = new Gpio(24, 'high');
const ps_wash_drain_switch = new Gpio(16, 'high');

module.exports.rodiOn = function() {
    rodiValve.writeSync(0);
    console.log("RODI water is on.");
}
module.exports.rodiOff = function() {
    rodiValve.writeSync(1);
    console.log("RODI water is off.");
}
```
calling `rodiOn` first, from the command line, closes all relays except `rodiValve`.
calling 'rodiOff` first closes all relays including the `rodiValve`.
calling `rodiOff immediatly after calling `rodiOn` closes the `rodiValve` relay. 

I'm not seeing this behavior with the Python program.

fivdi, 2018-06-13:
The JavaScript and Python programs are actually very different so I wouldn't expect them to behave in the same way.

Please read the documentation for the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) carefully, especially the description of the `direction` argument.

The JavaScript program is behaving in the way I would expect it to behave. What is the issue with the JavaScript program?

bmwertman, 2018-06-13:
I only brought up the Python program to show that they behave as expected running that program so I know my hardware, hardware configuration and wiring are good.

What I'm confused about is that I don't see anything in the documentation that tells me that running 'rodiValve.writeSync()', whether I pass it a 1 or a 0, should have any effect on my other configured GPIO.

fivdi, 2018-06-13:
> What I'm confused about is that I don't see anything in the documentation that tells me that running 'rodiValve.writeSync()', whether I pass it a 1 or a 0, should have any effect on my other configured GPIO.

It doesn't. But what does have an effect on other GPIOs is the following:
```js
const drain_relay = new Gpio(22, 'high');
const fill_relay = new Gpio(27, 'high');
const feed_relay = new Gpio(25, 'high');
const ps_wash_drain_switch = new Gpio(16, 'high');
```
Have you read the documentation for the Gpio constructor as suggested above and are you aware of how creation a Gpio object effects the value of the corresponding GPIO pin?

bmwertman, 2018-06-13:
I found the problem. Earlier when I was trying to solve my relay wiring issues I cam across issue https://github.com/fivdi/onoff/issues/57 which sounded similar to what I was seeing. I moved my GPIO config from `out` to `high` as you suggested. Changing it back to out fixed the problem.

I'm still not clear on the use case of `high` from the README.md. The only information I saw about `high` is 

> The valid values are: 'in', 'out', 'high', and 'low'. If 'out' is specified the GPIO will be configured as an output and the value of the GPIO will be set to 0. 'high' and 'low' are variants of 'out' that configure the GPIO as an output with an initial level of 1 or 0 respectively.

What does  ` "'high' and 'low' are variants of 'out' that configure the GPIO as an output with an initial level of 1 or 0 respectively"` mean? What is the use case where you wouldn't simply use `out`?