fivdi/onoff #114: Value of switch never changes.

thomaskblr, 2018-06-21:
How does one get the value of a switch upon start up? I tried button.read and button.readSync() and neither give the info needed.

fivdi, 2018-06-22:
The `read` and `readSync` methods are the correct methods to use.

Based on the information provided it's difficult to determine what the problem is.

- What type of board is being used?
- What is the number of the GPIO being used?
- What is the number of the pin on the expansion header being used?
- Exactly how is the button connected to the GPIO? A photo that shows everything clearly would help a lot here.
- Please post a complete program that is as short as possible that can be used to reproduce the problem.

fivdi, 2018-06-29:
@thomaskblr any news on this one?

thomaskblr, 2018-06-29:
Yes it was mistake on my end.  The documentation could have explained things a little better.  Thank you

fivdi, 2018-06-29:
Thanks for the feedback. A PR to improve the documentation or an issue describing what could be improved would be more than welcome.

closed 2018-06-29 by fivdi