fivdi/onoff #115: Support for active_low

splitice, 2018-06-26:
active_low is a relatively new property of the sysfs filesystem that allows the behaviour or Gpio's in an out direction to be flipped. This in particular changes the default state of the Gpio.

splitice, 2018-06-26:
Never mind I see there is support in the code.

fivdi, 2018-06-26:
> active low needs to be set first in order to invert the default value.

That makes sense. It also sounds like a bug in onoff. A potential workaround would be to set the mode to `high` rather than `out` (in addition to setting the constructors activeLow option to true.

splitice, 2018-06-26:
I'm not sure how that workaround would work, if you set the mode to ```low``` it would eventually set the state, however it would first go ```high``` and any device attached would detect the toggle.

splitice, 2018-06-26:
Another similar bug is that you should check the value of the ```direction``` before writing to it. Even if writing ```out``` when the direction is already ```out```. Writing to direction resets the ```value``` to the default (1) which causes issues if the pin is already exported.

fivdi, 2018-06-26:
> I'm not sure how that workaround would work, if you set the mode to low it would eventually set the state, however it would first go high and any device attached would detect the toggle.

You could be right. I haven't tested it and I only thought about the first run of the program after a reboot. If the program is terminated and run for a second time without rebooting with the GPIO still exported things will be different.

splitice, 2018-06-26:
it could look very similar to this: https://github.com/HalleyAssist/onoff/commit/4c042dd7fa13732081018048f5722d23f0443ad7

fivdi, 2018-06-26:
> Another similar bug is that you should check the value of the direction before writing to it. Even if writing out when the direction is already out. Writing to direction resets the value to the default (1) which causes issues if the pin is already exported.

This is what the documentation says should happen (apart from the activeLow bug.)

You have a use case where the state of the pin should not change if the pin is already exported. This isn't always the use case.

fivdi, 2018-06-26:
> @fivdi setting the value is why write exists. Setting the value is not a documented property or purpose of the constructor (nor should it be).

The constructor doesn't set the value of the GPIO explicitly. Configuring the GPIO as an output sets the value of the GPIO implicitly.

fivdi, 2018-07-23:
There are really two issues here.

1. onoff configures the GPIO `direction` before it configures `active_low` for a GPIO. It should be the other way around.

2. If a Gpio object is constructed onoff will always configure the GPIO `direction`. There are scenarios where this is not desirable. In some applications, if the `direction` is already configured correctly it should not be reconfigured. Reconfiguring an output GPIO as an output GPIO can result in the value of that GPIO changing. This reconfiguration could be avoided if the Gpio constructor supported an option which told it not to reconfigure the GPIO `direction` if it is already configured correctly.

I'll take a look at resolving both of these issues.

fivdi, 2018-07-24:
Both of the above issues have been resolved with onoff@3.2.0.

To resolve the second issue onoff@3.2.0 adds an new boolean option called reconfigureDirection to the Gpio constructor. This option can be used to retain the value of a GPIO output when an application is stopped and restarted.

closed 2018-07-24 by fivdi