fivdi/onoff #118: Update to new character device interface

markusr, 2018-09-12:
The sysfs interface for gpio in linux is deprecated, see [here](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt)

```
THIS ABI IS DEPRECATED, THE ABI DOCUMENTATION HAS BEEN MOVED TO
Documentation/ABI/obsolete/sysfs-gpio AND NEW USERSPACE CONSUMERS
ARE SUPPOSED TO USE THE CHARACTER DEVICE ABI. THIS OLD SYSFS ABI WILL
NOT BE DEVELOPED (NO NEW FEATURES), IT WILL JUST BE MAINTAINED.
```

Are there any plans to port this library to the new interface?
Or maybe create a new library using  [```libgpiod```](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/)?

fivdi, 2018-09-12:
There are currently no plans to port this library to the new interface. From a Node.js perspective using the libgpiod C library is more complex than using the GPIO sysfs interface as native bindings would be required to use libgpiod C library.

closed 2018-09-20 by fivdi

sombriks, 2020-12-29:
Would be nice to see libgpiod supported.
i've tried to write the node bindings myself with Nan to help with compatibility but no luck for the moment.

The maintainer of python-libgpiod commented something interesting on their codebase, he's trying to avoid to link with libgpiod directly, if i understood correctly, and trying a pure filesystem approach.

Maybe that's possible for onoff too.

fivdi, 2021-08-30:
> I'll try to make a high-quality native addon that exposes the newer chardev interface.

I wish you the best of luck working on this.

> ... sysfs may still be required on old systems, or users may want to avoid native addons. So it's nice for them to be able to choose.

It won't be possible to fully avoid native addons with onoff as onoff depends on the [epoll](https://github.com/fivdi/epoll) native addon for detecting the events generated by interrupt generating GPIO pins.