fivdi/onoff #119: Cannot mock static properties in my unit tests

uherj, 2018-10-17:
Hi,
I am trying to unit test my code that uses your Gpio library. I am unable to mock the static property accessible. That is the reason I cannot properly test all the code. Could you please somehow make it available? I am thinking of either through a getter (that is non-static so that I can mock the getter) or by making the property accessible non static?

Thank you.
Cheers,
Jarek

fivdi, 2018-10-17:
I'm sorry, but I don't want to make this change. I see it as a unit test issue rather than an onoff issue.

uherj, 2018-10-18:
My problem is that in the code I have
`
let led;

if (Gpio.accessible) {
   led = new Gpio(4, 'out');
} else {
    led = {
        writeSync: (value) => {
            console.log('virtual led now uses value: ' + value);
        }
    };
}
`
And my goal in the unit tests is to validate both paths. That both situations are correctly written. How should I do that?

Thank you.

fivdi, 2018-10-18:
Maybe by doing something like this:
```js
let myAccessible = Gpio.accessible;
let led;

if (myAccessible) {
   led = new Gpio(4, 'out');
} else {
    led = {
        writeSync: (value) => {
            console.log('virtual led now uses value: ' + value);
        }
    };
}
```
Gpio.accessible can't be modified but myAccessible can. This should make it possible to achieve what needs to be achieved.

Alternatively you could implement a Gpio mock.

johntalton, 2018-10-18:
if in a unit test you are mocking require ‘fs’ then the ‘accessable’ call will mock properly (as it’s juwt an fs try/catch)

the above code however looks like the use of accessible method to create a mock itself (which would be improper usage), otherwise, an in-app / in-test version of the above code should work unchanged.

the long term solution may be move to a Gpio factory/builder and only expose a common Gpio interface, and use dependency injection style.  this interface solution scales to support the current sysfs implementation as well as the new gpiochip interface and mocks etc (like mcp23xxx chip abstractions) … and move to typescript ...with added complexity

though, keeping this lib “simple” is likely the preferred outcome as this current sysfs interface is stable/well understood and added complexity does not serve this libs goals (that is, a “thin” wrapper over the legacy sysfs interface).

as a side note, that is a wider range of ‘accesssblity’ type calls that need to be added the the library to create they type of rich api needed to “understand” the sysfs interface from an abstract level.  and this i would suggest any serious discussion - of an interface change - to include and expand those concepts.

a further note, my understanding is that the sysfs gpio interface is legacy’s (as stated above) and that the new gpiochip interface is recommended.  again, i think this falls outside this libs goals as that requires the inclusion of a “native” module in order to access proper ioctrl calls ... again, among this lib a stable / legacy implementation that is best served by remaking simple at this point.