fivdi/onoff #121: watch only on certain pins?

coricidin12, 2018-12-10:
Does the watch function not work on all pins (pi 3b+)? I'm testing a motion sensor and it works fine on most GPIO pins using the watch function, but it's not detecting anything on pins 4, 5, and 6. Also doesn't detect anything on SDA (gpio2), SCL (gpio3), IDSC (gpio28), TXD (gpio14), RXD (gpio15), CE0 (gpio8), CE1 (gpio7). The other 17 pins work fine. Am I missing something in the setup?

fivdi, 2018-12-10:
Here is diagram of the GPIO header that I'll use for explaining things: https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png

Pin 3 or gpio2 (SDA) has an external 1.8kΩ pullup resistor soldered onto the Raspberry Pi. Does your circuit take this into account? If your circuit takes this into account watch will detect changes on this pin (it works for me.)
Pin 4 is a 5V power line and nothing can be detected here.
Pin 5 or gpio3 (SCL) has an external 1.8kΩ pullup resistor soldered onto the Raspberry Pi. Does your circuit take this into account? If your circuit takes this into account watch will detect changes on this pin (it works for me.)
Pin 6 is a ground line and nothing can be detected here.
Pin 28 or gpio0 (IDSC) (and not gpio28) should not be used, it's only for hats.

At this point I now think you are confusing physical pin numbers and GPIO numbers. onoff uses GPIO numbers and not physical pin numbers.

Also, if you have used the `raspi-config` tool to activate thinks like I2C, SPI or 1-Wire then the corresponding pins should only be used for I2C, SPI or 1-Wire respectively.

coricidin12, 2018-12-10:
Thanks for the response. When I said 4, 5, and 6, I meant gpio 4/5/6. All the pins I refer to above I meant gpio, not physical (except IDSC as 28, that was a mistake).

I didn't know about the pullups on gpio2/3, I'm sure that's the issue there. And IDSC being for hats explains that. So that leaves gpio 4/5/6/7(CE1)/8(CE0)/14(TXD)/15(RXD) still not working for me and without explaination. I2C, SPI, and 1-Wire are all disabled. Any other ideas?

fivdi, 2018-12-10:
Here's a link to another diagram of the GPIO header on a Raspberry Pi: http://exploringrpi.com/wp-content/uploads/2016/06/188681-c06f001.png

All of the GPIOs on the Raspberry Pi also have weak internal pullup and pulldown resistors (approx 50kΩ.) As the diagram shows, on some of the GPIOs the pullup resistors are enabled by default and on some of the GPIOs the pulldown resistors are enabled by default. With the exception of TXD and RXD the diagram shows that the GPIOs that you are having issues with all have pullup resistors enabled by default. It looks like your circuit doesn't work with weak (50kΩ) or strong (1.8kΩ) pullups.

I just tested with buttons connected to both TXD and RXD and watch detects changes when the buttons are pressed. This implies to me that it's most likely a circuit issue on your side here too. Note that on Raspbian a login shell is accessible over serial on TXD and RXD. Perhaps RXD and TXD will work for you if the login shell over serial is disabled with the `raspi-config` tool.

fivdi, 2018-12-10:
See also the [Configuring Pullup and Pulldown Resistors](https://github.com/fivdi/onoff#configuring-pullup-and-pulldown-resistors) section in the onoff readme.

coricidin12, 2018-12-11:
Turing off the pullups fixed the issue, thanks for your help.

fivdi, 2018-12-11:
You're welcome.

closed 2018-12-11 by fivdi