fivdi/onoff #123: Gpio constructor referencing BCM value instead of Gpio name

ThatsARapp, 2019-01-11:
Output of "gpio readall" for pin 7:
 | BCM | wPi  |   Name  | Mode | V | Physical |
 |   4     |   7 .  | GPIO. 7 |  OUT  | 1  |  7           |

"new Gpio(4, out)" creates an instance that references BCM 4 and Gpio 7 since BCM 4 equals gpio 7 on my setup.

How do I make the constructer reference the Gpio value instead of the BCM

Script:
light = new Gpio(4, out);
light.writeSync(0);
light.readSync(); //returns 0

After running script, and then typed in terminal:
gpio read 4 //returns 1. The value has not been updated
gpio read 7 //returns 0. The value has been updated

johntalton, 2019-01-11:
onoff is a sysfs wrapper and this uses numbering according to this legacy interface

the wiring pi gpio utility can be toggled to use bmc numbering via the -g flag

this makes sense and is consistent with the majority of documentation and general usage and insderstanding.

it may be a good idea to add a layer of constants to this library (aka BMC1…, PHY1… etc) to aid in clarity when constructing (or use a { physical: true } option, but that could get messy)

otherwise, any changes would break depending code

fivdi, 2019-01-11:
Duplicate of #86

fivdi, 2019-01-11:
I don't want to change onoff to enable it to use the Wiring Pi numbering scheme as the Wiring Pi numbering scheme is specific to the Raspberry Pi. onoff will run on any Linux platform. To be honest, I don't really understand why Wiring Pi introduced a new numbering scheme when a numbering scheme was already there.

fivdi, 2019-01-11:
> it may be a good idea to add a layer of constants to this library (aka BMC1…, PHY1… etc) to aid in clarity when constructing (or use a { physical: true } option, but that could get messy)

@johntalton Yes, this could get very messy and I don't think it would make things clearer.

As an example of how unclear things can be, I'm not actually 100% sure if the original post above is referring to physical pin 7 or Wiring Pi pin 7. In this case the numbers happen to be the same. Typically they are different, physical pin 40 is Wiring Pi pin 29 is BCM pin 21.

Things would start to get really messy with physical pin numbering on boards like the [BeagleBone Black](http://beagleboard.org/static/images/black_hardware_details.png) which have two GPIO headers.
```
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```