fivdi/onoff #125: activeLow pins initialize active

stuartpb, 2019-02-03:
The line `new Gpio(4, 'out', {activeLow: true})` appears to initialize GPIO 4 to 0, which is *active*. It should initialize it to 1.

stuartpb, 2019-02-03:
This doesn't look traceable directly to `onoff`'s initialization logic, so I'm still tracking the source of this issue down.

Repro steps: Run this script:

```js
const Gpio = require('onoff').Gpio;
const relay = new Gpio(4,'out',{activeLow: true});
```

Expected behavior: the state of GPIO 4 should not change, or, if it does change, it should be initialized high (meaning inactive).

Observed behavior: the state of GPIO 4 changes to low (active).

fivdi, 2019-02-03:
Which version of onoff is being used?

fivdi, 2019-02-03:
Note also that the values `'high'` and `'low'` can be passed to the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) as the `direction` instead of `'out'` to set the initial value of the GPIO to 1 or 0.

> While this should probably be fixed upstream

At this stage, I don't think the behaviour will ever be changed upstream.

stuartpb, 2019-02-03:
> Note also that the values `'high'` and `'low'` can be passed to the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) as the `direction` instead of `'out'` to set the initial value of the GPIO to 1 or 0.

This doesn't appear to be working correctly, at least not for GPIO pins that are set to activeLow.

fivdi, 2019-02-03:
> This doesn't appear to be working correctly, at least not for GPIO pins that are set to activeLow.

Does it do the opposite of what you expect? (because the kernel doesn't take activeLow into account when setting the direction)

stuartpb, 2019-02-03:
I see, `const relay = new Gpio(4, 'high', {activeLow: true});` is the one that will initialize an `activeLow` pin to the inactive state.

Is this documented, and if so, where?

fivdi, 2019-02-03:
> Is setting the direction to low to initialize implemented at the kernel level?

Yes.

> Is this documented, and if so, where?

See https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

fivdi, 2019-02-03:
It's also reasonably well document in the [onoff readme](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options).

fivdi, 2019-02-03:
I'm going to go ahead and close this issue as I don't think anything needs to be changed in onoff.

closed 2019-02-03 by fivdi