fivdi/onoff #126: Help needed on this subject: Switch on and off a lamp using relay

yvereg, 2019-02-11:
I designed a project for a heating regulation using node.js for the user interface. For some situations i need a pulse on the NO contact and i intend to achieve this using a relay connected to a GPIO port of a raspberry pi3. I followed the topic (https://github.com/fivdi/onoff/issues/102).
The relay is this one(https://fr.banggood.com/1-Channel-5V-Relay-Module-with-Optocoupler-Isolation-Relay-Board-p-1399429.html?akmClientCountry=FR&gmcCountry=FR&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=zouzou&utm_campaign=pla-frg-all-pc&cur_warehouse=CN) 
Unfortunatly, after the initializationthe relay is always on and therefore NO contact is always closed. I tried a lot of things without success.
I proceeded as suggested in the referred topic with no more success. In the right place I did:
echo GPIO > export / echo activeLow > 0 or 1/ echo out > direction / echo value 0 or 1  the relay is always activated after direction init and the attempts to change value has no effect.
Is there someone capable to tell me what is wrong in the operations I did?
Many thanks

7aman, 2019-02-11:
Hi @yvereg  
My Suggestions:  

- Please post your code and schematic.
- First make sure that your code is working. Replace relay with a simple "resistor + led" as load. Triggering some relays need more current than amount of current a raspberry pi can provide on its 5volt or gpio pins.

fivdi, 2019-02-11:
Did you try with
```js
const relay = new Gpio(<gpio-number>, 'high');
```
Or perhaps this:
```js
const relay = new Gpio(<gpio-number>, 'high', {reconfigureDirection: false});
```
For further information see the documentation for the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options).

yvereg, 2019-02-11:
hi @7aman Thanks for your answer.
Hereafter is the function I use to manage the relay, this function is called in the main program after pressing a button in a web page. Sorry for some comments in french.
function SetRelayOn(){
const Gpio = require ('onoff').Gpio;
//   if (defaut== "alarme.gif")  {
   if (defaut)  {
const Relay = new Gpio(4, 'out', activeLow = 1); //use GPIO pin 7, and specify that it is output
//console.log("1:",Relay.readSync());
       if(Relay.readSync()===0) { //si Gpio off, relay unclear
        Relay.writeSync(1);        //   Gpio on , relay activé
// console.log("2:",Relay.readSync());
      }
//remettre le relais au repos après 500ms
      setTimeout(function() {
      Relay.writeSync(0);  // Send 0 to switch off
      Relay.unexport();    // Clear pin state, free up resources
      }, 500);
   }	
//console.log("4:",Relay.readSync());
//console.log(defaut);
}

As you can see the unexport is done after 500ms, it is the mean i found to release the relay.

The code is working: I saw the console instruction when uncommented. Regardless to the sufficient power coming from the RPI, I assume it is ok because the relay is activated, the green led is lighting.
Thanks for your help
Regards

7aman, 2019-02-11:
You don't need to `require onoff` and declare `relay` in function's body.  
I suggest that you declare a `relay` at the top of your file. Also `unexport` the relay at the end of your file.  
for example:

```javascript
/* put this at the top of your file */
const Gpio = require('onoff').Gpio;
const relay = new Gpio(4, 'out', activeLow = 1);

/* rest of your code */

function setRelayOn(relay, duration) {
    relay.writeSync(1);
    setTimeout(function () {
        relay.writeSync(0);
    }, duration);
}

if (defaut) {
    setRelayOn(relay, 5*1000);
}

/* rest of your code */

/* put this at the end of your file */
process.on('SIGINT', () => {
    relay.unexport();
});
```

I increased duration from 500ms to 5 seconds to be sure it is working.  
I am not sure about `activeLow`. So maybe `relaySetOn` must be opposite of what I wrote above:

```javascript
function setRelayOn(relay, duration) {
    relay.writeSync(0);
    setTimeout(function () {
        relay.writeSync(1);
    }, duration);
}
```

fivdi, 2019-02-11:
The following code isn't using the Gpio constructors `activeLow` option correctly.
```js
const relay = new Gpio(4, 'out', activeLow = 1);
```
Please take a look at the readme to see how the `activeLow` option is used.

This issue is most likely a duplicate of https://github.com/fivdi/onoff/issues/125

yvereg, 2019-02-11:
@7aman, I aggree with you: I do'nt need to require onoff and declare relay in the body of the function. I do that only as a palliative. In the same way unexport is not in a good place , I put it there only to release the relay after the timer.  I will modify the code taking into account your remarks
@fivdi, I will follow your suggestions for the code modifications and I will explore the #125 subject. Thanks.
I will come back after my next session of tests
Regards

yvereg, 2019-02-11:
Hi, without activation of the SetRelayOn() function I checked all following initializations, between each the GPIO4 is released in a terminal session with the command /sys/class/gpio $ echo 4 > unexport

- const Relay = new Gpio(4, 'high');   relay on, green led on, NO contact closed
- const Relay = new Gpio(4, 'high', {reconfigureDirection: false}); relay on, green led on, NO contact closed
- const Relay = new Gpio(4, 'low');    relay on, green led on, NO contact closed
- const Relay = new Gpio(4, 'out', {activeLow : true}); relay on, green led on, NO contact closed
It reflects exactly what I said in my first message while I used only commands in a terminal session following instructions of the #102 subject. I will attentively read the post #125 trying to understand what is wrong in my approach.
Thanks

fivdi, 2019-02-11:
Can you post a photo of the circuit please?

fivdi, 2019-02-11:
Please post the contents of `/boot/config.txt` too. Perhaps there's a driver being used that also uses GPIO4.

yvereg, 2019-02-11:
hereafter the config.txt, followed by 2 pictures, DSC426 with relay on and DSC427 with relay off
 cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[image]
[image]

yvereg, 2019-02-12:
@7aman, regarding a part of your message:
P.S. after comment below, I removed activeLow. Set your relay to use active high by jumper on relay board.

For your information my relay board has no jumper to chooe the active level
regards

fivdi, 2019-02-12:
I was hoping the pictures would show which pins on the Raspberry Pi GPIO header and which pins on the relay were connected to each other. Unfortunately this is not the case so here are a few questions.

A picture of the Raspberry Pi GPIO header can be seen [here](https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png).

- Is one of the 5V power pins (and NOT one of the 3V3 power pins) on the Raspberry Pi GPIO header connected to the VCC pin on the relay?
- Is one of the Ground pins on the Raspberry Pi GPIO header connected to the GND pin on the relay?
- Is GPIO4 (pin 7) on the Raspberry Pi GPIO header connected to the IN pin on the relay?

I just tested a relay with the following program on a Raspberry Pi 3:
```js
const Gpio = require('onoff').Gpio;
const relay = new Gpio(4, 'high'); // IMPORTANT: Use 'high' if relay uses low level trigger

console.log('Program started. Relay should be off. Relay will turn on in 10 seconds.');

setInterval(() => {
  relay.writeSync(0);
  console.log('Relay should be on. Relay will turn off in 5 seconds.');

  setTimeout(() => {
    relay.writeSync(1);
    console.log('Relay should be off. ');
  }, 5000);
}, 10000);
```
The program works as expected turning the relay on for 5 seconds and then off for 5 seconds. It does this forever. The relay I'm using is a TONGLING relay just like your relay and has all the same markings as your relay. My relay is an active low relay meaning that setting the IN pin on the relay to 1 (high) will trun the relay off and setting the IN pin to 0 (low) will turn the relay on. This is why the Gpio object corresponding to the relay is created as follows to prevent the relay from turning on:
```js
const relay = new Gpio(4, 'high'); // IMPORTANT: Use 'high' if relay uses low level trigger
```
I don't know if your relay is an active low or an active high relay. Perhaps the page you link to above describing the relay says whether the relay is active low or active high. The page is French and unfortunately I don't speak French.

In order to determine whether your relay works I'd suggest trying to get it to work using only the sysfs files located at `/sys/class/gpio`. If it's not possible to get the relay to work using the sysfs files, then it will not work with onoff either. In this case there is some type of an electrical issue, for example, the relay is broken or the 5V power supply from the Raspberry Pi is not suitable for the relay and external 5V power supply is needed for the relay. Or perhaps some other electrical issue.

yvereg, 2019-02-12:
Hello, hereafter answers to your questions:
- GPIO pin 2 (5v) connected to VCC on relay board
- GPIO pin 6(GND) connected to GND on relay board
- GPIO pin 7(Gpio4) connected to IN on relay board 
As you suggest I used sysfs , and I made:
1st check with high
echo 4 > export /echo high > gpio4/direction , at this stage relay is on , the green led is on, value is 1. After echo 0 > gpio4/value the relay is still on and value is 0, end using echo 4 > unexport 
2nd check with low
echo 4 > export /echo low > gpio4/direction , at this stage relay is on , the green led is on, value is 0. 
After echo 1 > gpio4/value the relay is still on and value is 1, end using echo 4 > unexport 
3rd check with out  same behaviour as low

I bought this relay on amazon, i returned to see my command and  to see characteristics of it. It is not the same presentation, it 's an active low level whereas on the website in french for the tongling it is said that it  is an active high level. I saw comments and i'ts possible the relay has a problem. I will order another and I will come back .

many thanks
Regards

fivdi, 2019-02-15:
I'm going to go ahead and close this issue as the problem isn't related to onoff. Please feel free to add additional information to the issue after trying out the new relay.

closed 2019-02-15 by fivdi

yvereg, 2019-02-15:
OK, thanks for your help. I will try again in the coming weeks and i will
tell if something is new.