fivdi/onoff #129: Asynchronous write still sync?

patrickpissurno, 2019-02-14:
Hello,

First off, I'd like to thank you for this amazing library. It's the best around.

**The issue:**
I'm using the library with my Raspberry Pi 3b to make some music-playing floppy disks. I translated one program that I found online from C++ to Javascript and got Imperial March playing flawlessly on one floppy at a time.

The issue arrises when I try to play two of them simultaneously. The notes get strange and it seems like the commands are being interpolated.

Then I tried to spawn (with tmux) two node processes, one playing each floppy. This time I got the first floppy playing correctly, but the second one was mute until the first one stopped playing.

**Question:**
Is this something related to the Pi's GPIO being only able to handle one I/O at a time, or this is something related to the library itself?

This is my very first time working with GPIO, so I don't really know its limitations.

**Environment:**
Device: Raspberry Pi 3b
Distro: Raspbian GNU/Linux 9.3 (stretch)
Node version: v11.9.0
onoff version: 3.2.6

**Here is my code:** ([View on GitHub](https://github.com/patrickpissurno/flopperson/tree/6872dd216216e62aba7d13dfeaeb8d5dd0d79627))
```js
//MIT Licensed: view README.md
const { promisify } = require('util');
const { Gpio } = require('onoff');
const timer1 = new (require('nanotimer'))();
const timer2 = new (require('nanotimer'))();
const timestamp = new Date();

function sleep(ms) {
    // return new Promise(resolve => setTimeout(resolve, ms));
    return new Promise(resolve => timer1.setTimeout(resolve, '', ms + 'm'));
}
function microsleep(us) {
    return new Promise(resolve => timer2.setTimeout(resolve, '', us + 'u'));
}

function millis(){
    return new Date() - timestamp;
}

const {
    Cn, Cs, Dn, Ds, En, Fn, Fs, Gn, Gs, An, As, Bn, Zz,
    Bs, Df, Dn2, Ef, En2, Es, Gf, Gn2, Af, An2, Bf, Bn2, Zz2
} = require('./notes');

/**
 * Frequencies in hundredths of Hz, e.g. middle A = 44000
 * 4 Octaves with 12 notes per octave, i.e. C to B
 */
const freq = [
    [ 13081,13859,14683,15556,16481,17461,18500,19600,20765,22000,23308,24694 ],
    [ 26163,27718,29366,31113,32963,34923,36999,39200,41530,44000,46616,49388 ],
    [ 52325,55437,58733,62225,65925,69846,73999,78399,83061,88000,93233,98777 ],
    [ 104650,110873,117466,124451,131851,139691,147998,156798,166122,176000,186466,197553 ]
];

/**
 * Frequency (in Hz) is converted to Floppy Delay using the formula:
 *    314000 / frequency = floppy delay
 * so middle A = 314000 / 440 = 714
 *
 * Lowest realistic note is delay = 1550
 * Highest realistic note is delay = 210
 */
const floppyConv = 31400000;

// Calculate all our floppy delays at the start
const floppyDelay = JSON.parse(JSON.stringify(freq));
for (let octave = 0; octave < 4; octave++)
    for (let note = 0; note < 12; note++)
        floppyDelay[octave][note] = floppyConv / freq[octave][note];

class Floppy {
    constructor(directionPin, stepPin){
        this.directionPin = new Gpio(directionPin, 'out');
        this.stepPin = new Gpio(stepPin, 'out');
        this.directionPin.write = promisify(this.directionPin.write);
        this.stepPin.write = promisify(this.stepPin.write);

        this.resetMotor();
    }

    async resetMotor(){
        await this.directionPin.write(0);
        for(let i = 0; i < 10; i++){
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await sleep(1);
        }
        
        await this.directionPin.write(1);
        
        for(let i = 0; i < 10; i++){
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await sleep(1);
        }

        await sleep(400);
    }

    async playNote(note, octave, length){
        let dir = 1;
        const pause = floppyDelay[octave][note] * 10;
    
        const endTime = millis() + length;
        while (millis() < endTime){
            await this.directionPin.write(dir);
            
            //dir = 1 ? 0 : 1
            dir = 1 - dir;
    
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await microsleep(pause);
        }
    }

    async rest(length){
        const endTime = millis() + length;
        while (millis() < endTime)
            await sleep(5);
    }

    /**
     * song[note_num][note, octave, length]
     */
    async playSong(song, tempo)
    {
        // Convert tempo in BPM to millisecs
        const noteLen = 60000 / tempo;

        for (let i = 0; song[i][0] != -1; i++){
            const length = song[i][2] * noteLen;

            if (song[i][0] == Zz)
                await this.rest(length);
            else {
                await this.playNote(song[i][0], song[i][1], (7 * length) / 8);
                await this.rest(length / 8);
            }
        }
    }
}

const drive1 = new Floppy(17, 18);
const drive2 = new Floppy(22, 23);
const song = require('./hardcoded_songs/imperial_march');
drive1.playSong(song.notes, song.tempo);
drive2.playSong(song.notes, song.tempo);
```

fivdi, 2019-02-14:
> First off, I'd like to thank you for this amazing library. It's the best around.

Thank you very much :smile: 

I think the behaviour that you are seeing is related to how `NanoTimer` objects work. A `NanoTimer` object has a queue of timeouts rather than multiple timeouts running at the same time.

For example, the following program takes 15 seconds to run and not 5 seconds as might be expected.
```js
var NanoTimer = require('nanotimer');
var timer = new NanoTimer();

console.log('                 - ' + getTime());

function getTime() {
  return new Date().toISOString().substring(11, 19);
}

timer.setTimeout(function() {
  console.log('1s has expired   - ' + getTime());
}, '', '1s');

timer.setTimeout(function() {
  console.log('3s have expired  - ' + getTime());
}, '', '2s');

timer.setTimeout(function() {
  console.log('6s have expired  - ' + getTime());
}, '', '3s');

timer.setTimeout(function() {
  console.log('10s have expired - ' + getTime());
}, '', '4s');

timer.setTimeout(function() {
  console.log('15s have expired - ' + getTime());
}, '', '5s');
```
Hers's the output of the above program:
```
                 - 21:08:23
1s has expired   - 21:08:24
3s have expired  - 21:08:26
6s have expired  - 21:08:29
10s have expired - 21:08:33
15s have expired - 21:08:38
```
The example code posted in the original post creates two `Floppy` objects. Both of these `Floppy` objects use the same `NanoTimer` objects (timer1 and timer2) so they interfere with each other.

fivdi, 2019-02-14:
> I'm using the library with my Raspberry Pi 3b to make some music-playing floppy disks.

This sounds like a very interesting project.

patrickpissurno, 2019-02-15:
> This sounds like a very interesting project.

Thanks! Maybe take a look at some videos, if you like it. There are [some guys on YouTube](https://www.youtube.com/watch?v=q260bjSiyq0) doing the same.

**Got some progress**

After fixing that nanotimer thing, the interpolation issues are gone, but the results are now the same as spawning two node processes with tmux. Just like that situation I mentioned before:
> Then I tried to spawn (with tmux) two node processes, one playing each floppy. This time I got the first floppy playing correctly, but the second one was mute until the first one stopped playing.

Is there any limitation to writing simultaneously to many pins at a time? Because this code is basically flooding writes.

**Updated code:** ([View on GitHub](https://github.com/patrickpissurno/flopperson/tree/fd7b63aae057982bce07d2aa3e79f9713f08ca83))
```js
//MIT Licensed: view README.md
const { promisify } = require('util');
const { Gpio } = process.platform == 'linux' ? require('onoff') : { Gpio: function(){ this.write = (err, cb) => cb() } };
const nanotimer = require('nanotimer');
const timestamp = new Date();

function millis(){
    return new Date() - timestamp;
}

const {
    Cn, Cs, Dn, Ds, En, Fn, Fs, Gn, Gs, An, As, Bn, Zz,
    Bs, Df, Dn2, Ef, En2, Es, Gf, Gn2, Af, An2, Bf, Bn2, Zz2
} = require('./notes');

/**
 * Frequencies in hundredths of Hz, e.g. middle A = 44000
 * 4 Octaves with 12 notes per octave, i.e. C to B
 */
const freq = [
    [ 13081,13859,14683,15556,16481,17461,18500,19600,20765,22000,23308,24694 ],
    [ 26163,27718,29366,31113,32963,34923,36999,39200,41530,44000,46616,49388 ],
    [ 52325,55437,58733,62225,65925,69846,73999,78399,83061,88000,93233,98777 ],
    [ 104650,110873,117466,124451,131851,139691,147998,156798,166122,176000,186466,197553 ],
    [ 209300,221746,234931,248901,263702,279382,295995,313596,332243,352000,372931,395106 ]
];

/**
 * Frequency (in Hz) is converted to Floppy Delay using the formula:
 *    314000 / frequency = floppy delay
 * so middle A = 314000 / 440 = 714
 *
 * Lowest realistic note is delay = 1550
 * Highest realistic note is delay = 210
 */
const floppyConv = 31400000;

// Calculate all our floppy delays at the start
const floppyDelay = JSON.parse(JSON.stringify(freq));
for (let octave = 0; octave < freq.length; octave++)
    for (let note = 0; note < freq[0].length; note++)
        floppyDelay[octave][note] = floppyConv / freq[octave][note];

const notesNames = [
    'C', 'C#', 'D', 'D#', 'E', 'F', 'F#', 'G', 'G#', 'A', 'A#', 'B'
];

class Floppy {
    constructor(directionPin, stepPin){
        this.directionPin = new Gpio(directionPin, 'out');
        this.stepPin = new Gpio(stepPin, 'out');
        this.directionPin.write = promisify(this.directionPin.write);
        this.stepPin.write = promisify(this.stepPin.write);

        this._timer1 = new nanotimer();
        this._timer2 = new nanotimer();

        this.sleep = (ms) => {
            // return new Promise(resolve => setTimeout(resolve, ms));
            return new Promise(resolve => this._timer1.setTimeout(resolve, '', ms + 'm'));
        };
        this.microsleep = (us) => {
            return new Promise(resolve => this._timer2.setTimeout(resolve, '', us + 'u'));
        };

        this.resetMotor();
    }

    async resetMotor(){
        await this.directionPin.write(0);
        for(let i = 0; i < 10; i++){
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await this.sleep(1);
        }
        
        await this.directionPin.write(1);
        
        for(let i = 0; i < 10; i++){
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await this.sleep(1);
        }

        await this.sleep(400);
    }

    async playNote(note, octave, length){
        console.log('Note: ' + notesNames[note] + (octave + 1));   

        let dir = 1;
        const pause = floppyDelay[octave][note] * 10;
    
        const endTime = millis() + length;
        while (millis() < endTime){
            await this.directionPin.write(dir);
            
            //dir = 1 ? 0 : 1
            dir = 1 - dir;
    
            await this.stepPin.write(1);
            await this.stepPin.write(0);
            await this.microsleep(pause);
        }
    }

    async rest(length){
        const endTime = millis() + length;
        while (millis() < endTime)
            await this.sleep(5);
    }

    /**
     * song[note_num][note, octave, length]
     */
    async playSong(song, tempo)
    {
        // Convert tempo in BPM to millisecs
        const noteLen = 60000 / tempo;

        for (let i = 0; song[i][0] != -1; i++){
            const length = song[i][2] * noteLen;

            if (song[i][0] == Zz)
                await this.rest(length);
            else {
                await this.playNote(song[i][0], song[i][1], (7 * length) / 8);
                await this.rest(length / 8);
            }
        }
    }
}

module.exports = Floppy;
```

fivdi, 2019-02-15:
> Is there any limitation to writing simultaneously to many pins at a time? Because this code is basically flooding writes.

I'm not aware of any limitations. That being said, I'm not familiar with the implementation details of the Linux drivers and can't make any statement about the exact details of how they function.

> After fixing that nanotimer thing, the interpolation issues are gone, but the results are now the same as spawning two node processes with tmux. Just like that situation I mentioned before:

>> Then I tried to spawn (with tmux) two node processes, one playing each floppy. This time I got the first floppy playing correctly, but the second one was mute until the first one stopped playing.

I gave the above program a try with LEDs rather than floppies. The LEDs blinked together in the same pattern most of the time but not all the time. The second LED wasn't off till the first one stopped. To be honest, I wouldn't expect to see the exact pattern on both LEDs all the time because I don't think it's going to be possible to get independent timers to remain fully synchronous with one another using JavaScript running on Raspbian.

Here's what was displayed on the screen
```
Note: G2
Note: G2
Note: G2
Note: G2
Note: G2
Note: G2
Note: D#2
Note: D#2
Note: A#2
Note: A#2
Note: G2
Note: G2
Note: D#2
Note: D#2
Note: A#2
Note: A#2
Note: G2
Note: G2
Note: D3
Note: D3
Note: D3
Note: D3
Note: D3
Note: D3
Note: D#3
Note: D#3
Note: A#2
Note: A#2
Note: F#2
Note: F#2
Note: D#2
Note: D#2
Note: A#2
Note: A#2
Note: G2
Note: G2
Note: G3
Note: G3
Note: G2
Note: G2
Note: G2
Note: G2
Note: G3
Note: G3
Note: F#3
Note: F#3
Note: F3
Note: F3
Note: E3
Note: E3
Note: D#3
Note: D#3
Note: E3
Note: E3
Note: G#2
Note: G#2
Note: C#3
Note: C#3
Note: C3
Note: C3
Note: B2
Note: B2
Note: A#2
Note: A#2
Note: A2
Note: A2
Note: A#2
Note: A#2
Note: D#2
Note: D#2
Note: F#2
Note: F#2
Note: D#2
Note: D#2
Note: F#2
Note: F#2
Note: A#2
Note: A#2
Note: G2
Note: G2
Note: A#2
Note: A#2
Note: D3
Note: D3
Note: G3
Note: G3
Note: G2
Note: G2
Note: G2
Note: G2
Note: G3
Note: G3
Note: F#3
Note: F#3
Note: F3
Note: F3
Note: E3
Note: E3
Note: D#3
Note: D#3
Note: E3
Note: E3
Note: G#2
Note: G#2
Note: C#3
Note: C#3
Note: C3
Note: C3
Note: B2
Note: B2
Note: A#2
Note: A#2
Note: A2
Note: A2
Note: A#2
Note: A#2
Note: D#2
Note: D#2
Note: F#2
Note: F#2
Note: D#2
Note: D#2
Note: A#2
Note: A#2
Note: G2
Note: G2
Note: D#2
Note: D#2
Note: A#2
Note: A#2
Note: G2
Note: G2
```

fivdi, 2019-02-17:
@patrickpissurno I'm going to go ahead and close this issue as there doesn't appear to be a problem with onoff. If it turns out that there is a problem with onoff the issue can be reopened. Feel free to comment on this issue if further information becomes available.

closed 2019-02-17 by fivdi