fivdi/onoff #131: Error: EACCES: permission denied errors

emilienh, 2019-02-21:
Hello,
I am getting `Error: EACCES: permission denied, open '/sys/class/gpio/gpio16/value'` errors the first few times I try using the `new Gpio()` constructor when using version 3.2.7 of this library. I noticed that the `waitForAccessPermission` method used to try to open the gpio file 10 000 times before giving up, and that number was reduced to 1 000 tries in the latest version. It looks like that's not enough in our case. 

onoff 3.2.7
Node 8.13.0
Raspberry Pi 3B+
Poky 2.5.2 - Kernel 4.14.68

guillaumekh, 2019-02-21:
https://github.com/fivdi/onoff/blob/454598f016f1a37f032e5f3bd9fb6f1e4412a34a/onoff.js#L28

fivdi, 2019-02-21:
@emilienh Thank you for reporting this issue and sorry for any inconvenience caused. I'll change number of attempts that `waitForAccessPermission` makes to access files from 1 000 back to 10 000.

> https://github.com/fivdi/onoff/blob/454598f016f1a37f032e5f3bd9fb6f1e4412a34a/onoff.js#L28

> What was the rationale behind lowering the max number of retries ?

@guillaumekh The modification was made to make a unit test pass without timing out. The tests that I performed with real hardware after decreasing the number of tries from 10 000 to 1 000 didn't have any EACCES issues.

fivdi, 2019-02-21:
Fixed by 6a6e177c1a1637447be6cc361ccdfdf021173a75 and published on npm with onoff@3.2.8.

closed 2019-02-21 by fivdi