fivdi/onoff #133: Not able to produce output at all

saenglert, 2019-02-24:
Hey everyone,
I've recently added a [2.2" display with 4 buttons](https://learn.adafruit.com/adafruit-2-2-pitft-hat-320-240-primary-display-for-raspberry-pi/overview) to a Pi 2 B and I'm not able to get any inputs from the buttons. The display/buttons are definitly working as I can read them just fine via Python.

My setup:
```
{
  "name": "nodegpio",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "start": "node ./src/index.js"
  },
  "author": "",
  "license": "MIT",
  "dependencies": {
    "onoff": "^3.2.8"
  }
}
```

My test script:
```
const Gpio = require('onoff').Gpio;
const buttons = [
    new Gpio(17, 'in', 'both'),
    new Gpio(22, 'in', 'both'),
    new Gpio(23, 'in', 'both'),
    new Gpio(27, 'in', 'both'),
];
 
buttons.forEach(button => button.watch((err, value) => {
        console.log("Button pressed");
        if (err) { 
            throw err; 
        }

        console.log(value)
    })
)

process.on('SIGINT', () => {
    buttons.forEach(button => {
        button.unexport();
    });
    console.log("exiting");
});
```

fivdi, 2019-02-24:
The PiTFT [circuit diagram](https://cdn-learn.adafruit.com/assets/assets/000/021/903/original/adafruit_products_schem.png?1419022497) shows that the four buttons don't have any external pullup or pulldown resistors. When one of the buttons is pressed, the corresponding GPIO is simply pulled down to 0V or pulled LOW.

On the other hand, page 102 of the [BCM2835 ARM Peripherals Documentation](http://www.farnell.com/datasheets/1521578.pdf) shows that GPIOs 17, 22, 23 and 27 all have their internal pulldown resistors enabled by default. This means that when a button is not pressed it will also read as 0V or LOW.

In other words, the buttons will always read as 0V or LOW irrespective of whether they are pressed or not. This in turn means that the program posted above will never detect any button state changes.

To get it to work, the internal pullup resistors on the four GPIOs connected to the four buttons should be enabled. This can be achieved by adding the following line at the end of file  `/boot/config.txt` on the Raspberry Pi and rebooting the Raspberry Pi.
```
gpio=17,22,23,27=pu
```
Please give this a try to see if it works. For further information see the [Configuring Pullup and Pulldown Resistors](https://github.com/fivdi/onoff#configuring-pullup-and-pulldown-resistors) section of the readme.

saenglert, 2019-02-24:
Works like a charm.
Thank you very much for taking a look at this and for the detailed explanation!

closed 2019-02-24 by saenglert

fivdi, 2019-02-24:
Excellent, good to hear that it's working now thank you for the feedback.