fivdi/onoff #137: Callback parameters not following best practice

saenglert, 2019-03-02:
At the moment the a callback for `read` and `watch` (and by extend `write` as well) has to look like this:
```
(err: Error | undefined, value: 1 | 0) => void
```
`err` has to be an Error or undefined and `value` is either `1` or `0`. Optional parameters should be listed last as to not interfere with required parameters as well as to be able to omit them entirely:

```
const callback1 = (err: Error | undefined, value: 1 | 0) => {}
callback1(new Error("Hello World", 0) // works
callback1(undefined, 0) // works
callback1(0) // fails

const callback2 = ( value: 1 | 0, err?:Error) => {}

const callback2(0, undefined) // works
callback2(0, new Error("Hello World")) // works
callback2(0) // works

// Note that
value?: number
// is not the same as 
value: number | undefined
// The first example can be omitted while for the second value either a number or undefined has to be passed to it.
```
In the first example `err` can't be omited and a value has to be passed because otherwise it would mess with the order of the following parameters.

The only downside to this approach now if an error occures a value for `value` has to be passed down to avoid the same scenario just turned 180. In this case I'd suggest handing down an error value (In the current case for example `-1`). But since the first thing one does in a callback is the check wether or not the `err` is defined or not the value is motlikely discarded anyway.

fivdi, 2019-03-02:
The asynchronous methods in onoff follow the "error-first callback" idiomatic pattern that most asynchronous methods in the Node.js core API also follow. There's a short description of the pattern [here](https://nodejs.org/api/errors.html#errors_error_first_callbacks).

When onoff was originally implemented the description for the "error-first callback" idiomatic pattern that I used at the time was [this one](https://nodejs.org/dist/latest-v11.x/docs/api/fs.html#fs_file_system):

> All file system operations have synchronous and asynchronous forms.

> The asynchronous form always takes a completion callback as its last argument. The arguments passed to the completion callback depend on the method, but the first argument is always reserved for an exception. If the operation was completed successfully, then the first argument will be null or undefined.

Note that the `err` argument isn't optional when the "error-first callback" pattern is used. If the operation completes successfully the `err` argument is `null` or `undefined`.

saenglert, 2019-03-02:
I guess we are coming from different backgrounds on this. From a TypeScript perspective this sort of handling is discuraged because of the build in `?` syntax. Since it's your project and you obviously put some thought into it I'm going to work with it.

closed 2019-03-02 by saenglert