fivdi/onoff #143: fs.existsSync is not a function

gecevicius, 2019-03-17:
Hello,
Working on a Node + onoff project. Trying to create a simple LED function, but I am receiving an error that I can't find a way to fix. my code : 
`var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var LED = new Gpio(4, 'out'); //use GPIO pin 4, and specify that it is output
var blinkInterval = setInterval(blinkLED, 250); //run the blinkLED function every 250ms

function blinkLED() { //function to start blinking
  if (LED.readSync() === 0) { //check the pin state, if the state is 0 (or off)
    LED.writeSync(1); //set pin state to 1 (turn LED on)
  } else {
    LED.writeSync(0); //set pin state to 0 (turn LED off)
  }
}

function endBlink() { //function to stop blinking
  clearInterval(blinkInterval); // Stop blink intervals
  LED.writeSync(0); // Turn LED off
  LED.unexport(); // Unexport GPIO to free resources
}

setTimeout(endBlink, 5000); //stop blinking after 5 seconds`

Error : 
`TypeError: fs.existsSync is not a function
    at exportGpio (onoff.js?b64a:16)
    at new Gpio (onoff.js?b64a:172)`

Node version 10.10.0. The LED works fine with a python script.

fivdi, 2019-03-17:
I just tried the following program with Node v10.10.0 and it works fine for me. 
```js
var Gpio = require('onoff').Gpio;
var LED = new Gpio(17, 'out'); //use GPIO pin 4, and specify that it is output
var blinkInterval = setInterval(blinkLED, 250); //run the blinkLED function every 250ms
function blinkLED() { //function to start blinking
  if (LED.readSync() === 0) { //check the pin state, if the state is 0 (or off)
    LED.writeSync(1); //set pin state to 1 (turn LED on)
  } else {
    LED.writeSync(0); //set pin state to 0 (turn LED off)
  }
}
function endBlink() { //function to stop blinking
  clearInterval(blinkInterval); // Stop blink intervals
  LED.writeSync(0); // Turn LED off
  LED.unexport(); // Unexport GPIO to free resources
}
setTimeout(endBlink, 5000); //stop blinking after 5 seconds
```
The LED blinks for 5 seconds and then the program gracefully terminates as is to be expected. Two modifications were made to the program. GPIO17 was used in place of GPIO4 and the following line was added to the program:
```js
var Gpio = require('onoff').Gpio;
```
How come this line was missing from the program you posted?
Is anything else missing from the code you posted?
Is there something special about the way `require` is handled on your system?

What is the output of the following program on your system:
```js
var fs = require('fs');
console.log(fs.existsSync);
```
Hers's the output that I see with Node v10.10.0 on a Raspberry Pi:
```
[Function: existsSync]
```

gecevicius, 2019-03-17:
Sorry, left that line out accidentally. I tried it with a few GPIO pin numbers, 4 and 17 included. Printing fs.existsSync gave me _'undefined'_ . Everything is pretty much stock settings in terms of webpack. There is something wrong in my app since I have the same exact output on my Windows machine as well as the Pi, not sure what.

fivdi, 2019-03-17:
> Printing fs.existsSync gave me 'undefined'

Are you running the program from the command line using a command like `node yorrAppName.js`? If not, I wouldn't expect it to work.

> Everything is pretty much stock settings in terms of webpack.

What is webpack being used for? I thought webpack was for bundling JavaScript files for usage in a browser. Why is it being used here? Note that `onoff` is not suitable for usage in a browser.

gecevicius, 2019-03-17:
Oh that's the problem then. I was not aware it does not support browsers. Apologies! Could you point me to the right direction then? Thanks!

fivdi, 2019-03-17:
I'd suggest taking a look at a tutorial about using Node.js in general or using Node.js on a Raspberry Pi. For example, [this tutorial](https://www.w3schools.com/nodejs/nodejs_raspberrypi.asp) explains how to use Node.js and onoff on a Raspberry Pi.

I'm going to go ahead and close this issue as there doesn't appear to be a problem with onoff.

closed 2019-03-17 by fivdi