fivdi/onoff #145: Run without epoll?

lordbah, 2019-05-06:
If I'm only using GPIO for output, is it possible to run without epoll? I'm in a situation where I'm on amd64 Windows building a docker container which will run on armv71 Pi. At runtime that gets

`... epoll.node: invalid ELF header`

even doing `build --platform arm`

If I don't need epoll, is there a way to exclude it at build time?
If I do need it, I guess I'm looking at building a docker container whose entrypoint does an
`npm install onoff`
and which contains a compiler??

fivdi, 2019-05-07:
> If I don't need epoll, is there a way to exclude it at build time?

I'm afraid there isn't a way to exclude epoll at build time.

>If I do need it, I guess I'm looking at building a docker container whose entrypoint does an
npm install onoff
>and which contains a compiler??

I'm not really familiar with docker and can't provide any further information here.

fivdi, 2019-05-11:
There doesn't appear to be a todo for onoff here so I'm going to go ahead and close the issue.

closed 2019-05-11 by fivdi