fivdi/onoff #146: issue on power failure

studiosacchetti, 2019-05-07:
i'm using this library on raspy bi 3+ (jessie) and i found  that some time library doesn't work, especially on a power failure.

my workaround at the momnent is lauch this python script before nodejs app and everything goes fine

```
import RPi.GPIO as GPIO
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BOARD)
GPIO.setup(7, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
```
**hardware** 

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000ceae70cb
----------------

fivdi, 2019-05-07:
In order for the button to function correctly the internal pulldown resistor on the corresponding GPIO needs be enabled. This can be achieved by adding the following line at the end of file  `/boot/config.txt` on the Raspberry Pi and rebooting the Raspberry Pi.
```
gpio=4=pd
```
Please give this a try to see if it works. For further information see the [Configuring Pullup and Pulldown Resistors](https://github.com/fivdi/onoff#configuring-pullup-and-pulldown-resistors) section of the readme.

Note that onoff uses GPIO numbers rather than pin numbers. Pin 7 is GPIO 4 so the number to use in onoff is 4.

studiosacchetti, 2019-05-08:
nope, doesen't work. 
and yes i use gpio4 in node js

also a simple code like this wont work

```
const Gpio = require('onoff').Gpio;
const button = new Gpio(4, 'in', 'both');

button.watch((err, value) => {
  console.log('value:' +value) 
});

process.on('SIGINT', () => {
  button.unexport();
});
```

fivdi, 2019-05-08:
Does it work if an external 1k pullup resistor is used for the button as shown in the [usage section]( https://github.com/fivdi/onoff#usage) of the readme?

studiosacchetti, 2019-05-10:
i'm actually using a different resistor, do you suggest to change it with a 10k?

fivdi, 2019-05-10:
What value does the resistor that is being used have?

studiosacchetti, 2019-05-10:
should be 100 ohm

fivdi, 2019-05-11:
>should be 100 ohm

This is a very vague answer.

If it is a 100 ohm resistor that would be a very strong pulldown resistor so it should work.

The Python program enables the internal weak pulldown resistor and I don't understand how running the Python program can change anything as there is already a strong pulldown resistor.

When the Node.js program is running and the button is not being pressed, run the command `cat /sys/class/gpio/gpio4/value`. It should print `0`. Does it?

When the Node.js program is running and the button is held pressed, run the command `cat /sys/class/gpio/gpio4/value`. It should print `1`. Does it?

I don't think the issue is related to onoff.

studiosacchetti, 2019-05-13:
**with  py init script before**

button not pressed
`cat /sys/class/gpio/gpio4/value  = 0`

pressing button

`cat /sys/class/gpio/gpio4/value  = 1`

**without py ninit script before** 
after power failure no button pressed
`cat /sys/class/gpio/gpio4/value  = 1`

pressing button
`cat /sys/class/gpio/gpio4/value  = 1`

fivdi, 2019-05-13:
This is so strange. I wonder if we'll ever figure out what the issue is...

- Please post the complete content of the Raspberry Pi file `/boot/config.txt`
- Does the directory `/sys/class/gpio/gpio4` exist directly after booting the Raspberry Pi and before running the Node.js program or Python script for the first time?
- After a reboot, when the Node.js program is running and without having ever run the Python script, run the command `gpio readall` in a second terminal window and post the complete output of that command.
- After a reboot, when the Node.js program is running and without having ever run the Python script, run the command `cat /sys/class/gpio/gpio4/direction` in a second terminal window and post the output of that command.
- Please post the output of command `uname -a`
- What version of onoff is being used?
- What version of Node.js is being used? This can be determined with the command `node --version`
- Is sudo being used? It should not be necessary to used sudo.
- Are any other programs that use GPIOs being run on the Raspberry Pi at the same time as the Node.js program?

fivdi, 2019-05-13:
Just to be sure, the following Node.js program is being used for testing, correct?
```js
const Gpio = require('onoff').Gpio;
const button = new Gpio(4, 'in', 'both');

button.watch((err, value) => {
  console.log('value:' +value) 
});

process.on('SIGINT', () => {
  button.unexport();
});
```

studiosacchetti, 2019-05-15:
```
# For more options and information see
# http://rpf.io/configtxtreadme
# Some settings may impact device functionality. See link above for details
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1
# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1
# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16
# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720
# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1
# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=81
# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2
# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4
# uncomment for composite PAL
#sdtv_mode=2
#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi
# Additional overlays and parameters are documented /boot/overlays/README
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=pi3-disable-bt
enable_uart=0
#display_rotate 3
gpu_mem=64
```

studiosacchetti, 2019-05-15:
- Does the directory /sys/class/gpio/gpio4 exist directly after booting the Raspberry Pi and before running the Node.js program or Python script for the first time?  no
```
-rwxrwx---  1 root gpio 4096 mag 15 14:27 export
lrwxrwxrwx  1 root gpio    0 mag 15 14:27 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root gpio    0 mag 15 14:27 gpiochip504 -> ../../devices/platform/soc/soc:firmware/soc:firmware:expgpio/gpio/gpiochip504
-rwxrwx---  1 root gpio 4096 mag 15 14:27 unexport

```
- After a reboot, when the Node.js program is running and without having ever run the Python script, run the command gpio readall in a second terminal window and post the complete output of that command.
 ```
+-----+-----+---------+------+---+---Pi 3+--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 0 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |  OUT | 0 | 11 || 12 | 1 | OUT  | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |  OUT | 0 | 15 || 16 | 1 | OUT  | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | OUT  | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 1 | OUT  | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3+--+---+------+---------+-----+-----+
```

After a reboot, when the Node.js program is running and without having ever run the Python script, run the command cat /sys/class/gpio/gpio4/direction in a second terminal window and post the output of that command.
Please post the output of command uname -a
What version of onoff is being used?
What version of Node.js is being used? This can be determined with the command node --version
Is sudo being used? It should not be necessary to used sudo.
Are any other programs that use GPIOs being run on the Raspberry Pi at the same time as the Node.js program?

fivdi, 2019-05-18:
>After a reboot, when the Node.js program is running and without having ever run the Python script, run the command gpio readall in a second terminal window and post the complete output of that command.
```
+-----+-----+---------+------+---+---Pi 3+--+---+------+---------+-----+-----+
| BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
+-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
|     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
|   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
|   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
|   4 |   7 | GPIO. 7 |   IN | 0 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
|     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
|  17 |   0 | GPIO. 0 |  OUT | 0 | 11 || 12 | 1 | OUT  | GPIO. 1 | 1   | 18  |
|  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
|  22 |   3 | GPIO. 3 |  OUT | 0 | 15 || 16 | 1 | OUT  | GPIO. 4 | 4   | 23  |
|     |     |    3.3v |      |   | 17 || 18 | 0 | OUT  | GPIO. 5 | 5   | 24  |
|  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
|   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 1 | OUT  | GPIO. 6 | 6   | 25  |
|  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
|     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
|   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
|   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
|   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
|  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
|  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
|  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
|     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
+-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
| BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
+-----+-----+---------+------+---+---Pi 3+--+---+------+---------+-----+-----+
```
>Are any other programs that use GPIOs being run on the Raspberry Pi at the same time as the Node.js program?
no

If no other programs are using GPIOs why does the output of the command `gpio readall` say that GPIO17, GPIO18, GPIO22, GPIO23, GPIO24 and GPIO25 are configured to Mode "OUT"? What configured them to Mode "OUT". If nothing was using them they should be configured to Mode "IN" by default after a reboot.

I probably should have asked this question before, but what exactly do you mean when you say power failure in the above posts? How come there are so many power failures?

>Is sudo being used? It should not be necessary to used sudo.
it run like root

Why is it necessary to use root? If user `pi` is being used on Raspbian it should not be necessary to use sudo.

Is the Raspbian operting system being used?

Can you post a photo of the circuit please? The photo should shown the button, resistors being used, wiring, and Raspberry Pi GPIO header.

studiosacchetti, 2019-05-20:
`> 9f no other programs are using GPIOs why does the output of the command gpio readall say that GPIO17, GPIO18, GPIO22, GPIO23, GPIO24 and GPIO25 are configured to Mode "OUT"? What configured them to Mode "OUT". If nothing was using them they should be configured to Mode "IN" by default after a reboot.`
it's because also a lcd tisplay is connected, but problem occur even without lcd (lcd work ok)

I probably should have asked this question before, but what exactly do you mean when you say power failure in the above posts? How come there are so many power failures?
removed power plug from socket 

> 
>     Is sudo being used? It should not be necessary to used sudo.
>     it run like root
> Why is it necessary to use root? If user pi is being used on Raspbian it should not be necessary to use sudo.
it because ai created a serice in /etc/serviced, but happen the same if i run manually from pi user

> Is the Raspbian operting system being used?
yes of course

> Can you post a photo of the circuit please? The photo should shown the button, resistors being used, wiring, and Raspberry Pi GPIO header.
ok dou you want to connect it to raspy with teamviewer?

studiosacchetti, 2019-05-20:
[image]

fivdi, 2019-05-20:
Observations related to the button:

- One of the pins on the left-hand side of the button is connected to GPIO4 via the yellow wire which is correct.
- The right-hand side of the button appears to be connected to a screw, is this correct? If so, why?
- There is a red wire connecting the 3.3V pin on the Raspberry Pi GPIO header to the board with the button. Is this red wire connected to one of the pins on the right-hand side of the button?
- I can't see all colored bands on the resistor. What color are the bands?
- I can't see what both ends of the resistor is connected to.
- What are both ends of the resistor connected to?
- I can't see any connection between 0V on the Raspberry Pi and anything on the button board so there doesn't appear to be a pulldown resistor on the button board.
- Do you know what a pulldown resistor is?
- If you post a picture of the rear side of the button board I could see what is connected to what.

> removed power plug from socket, end user normally reset like that

Simply pulling out the plug from the socket isn't a good idea. Linux should be shutdown correctly first.

In order to move forward here I think it will be easier to go back to square one and start again. Please flash an SD card with Raspbian and don't modify the Raspbian installation. Leave it the way it is. Don't create the service in /etc/serviced. Install node if necessary. Then try to get the button to work.

If the resistor on the button board is removed and the yellow wire is connected to the left-hand side of the button (as is already the case) and the red wire is connected to the right hand-side of the button and the following line is added to `/boot/config.txt` the button should work:
```
gpio=4=pd
```
Try this first to see if it's possible to get the button to work.

Aneesawan34, 2019-05-23:
When I run this program I get this error 
Const high_buff = buffer.from(1)

Type Error: this is not a typed array

studiosacchetti, 2019-05-23:
i'm investingating on button it might be a problem of that. I'm not that into electronic and circuits..
i will keep u updated

fivdi, 2019-05-23:
> i'm investingating on button it might be a problem of that. I'm not that into electronic and circuits..
i will keep u updated

@studiosacchetti Ok, no problem.

@Aneesawan34 Please open a new issue describing the problem you are having. Please include a complete program that can be used to reproduce the problem in the new issue.

fivdi, 2019-06-08:
@studiosacchetti Was it possible to make any progress here?

fivdi, 2019-06-12:
@studiosacchetti I'm going to go ahead and close this one. If it turns out to be an issue with onoff the issue can of course be reopened.

closed 2019-06-12 by fivdi