fivdi/onoff #147: Don't set direction twice

benwaffle, 2019-05-20:
If `direction` is `out`, and `value` is 1, and you run `echo out > direction`, value is reset to 0. The constructor for `Gpio` should check if the direction actually needs to be written.

fivdi, 2019-05-20:
As mentioned in the readme, this can be achieved by setting the [constructors](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) `reconfigureDirection` option to false.

benwaffle, 2019-05-20:
Also turns out I can use `high` and `low` in the direction field to atomically set the direction and value

closed 2019-05-20 by benwaffle

fivdi, 2019-05-20:
> Also turns out I can use high and low in the direction field to atomically set the direction and value (Linux does this).

Correct. This is also mentioned in the readme.