fivdi/onoff #150: node 12 fs.write is enforcing callback must be supplied

sdetweil, 2019-06-11:
latest node fs library is enforcing callbacks on async write..

Gpio.prototype.write

should be changed to match like read 
```
Gpio.prototype.write = function (value, callback) {
  var writeBuffer = value === 1 ? ONE : ZERO;
  fs.write(this.valueFd, writeBuffer, 0, writeBuffer.length, 0, function (err, bytes, buf) {
    if (typeof callback === 'function') {
      if (err) {
        return callback(err);
      }
      callback(null, bytes, buf);
    }
  });
};
```

hover-nodejs failure 
TypeError [ERR_INVALID_CALLBACK]: Callback must be a function
at maybeCallback (fs.js:157:9)
at Object.fs.write (fs.js:652:16)
at Gpio.write (/home/pi/MagicMirror/modules/MMM-Hover/node_modules/onoff/onoff.js:202:6)
at Timeout._onTimeout (/home/pi/MagicMirror/modules/MMM-Hover/node_modules/hover-nodejs/Hover.js:50:15)

fivdi, 2019-06-11:
Updating to the latest version of on off (v4.1.1) should resolve this issue.

sdetweil, 2019-06-11:
ok, hover-nodejs requested ^1.1.1, got 1.2.0

fivdi, 2019-06-11:
In that case hover-nodejs would need to upgrade to onoff@4.1.1. Or you could downgrade to an older version of Node.js. In either case, I'm afraid there is nothing that onoff can do to resolve the issue.

sdetweil, 2019-06-11:
understood... closing now... thanks for the prompt info..

closed 2019-06-11 by sdetweil