fivdi/onoff #152: Watch continuously triggers even though the value hasn't changed, only after 8hr+

smysnk, 2019-06-13:
I have been developing a hydroponics controller of sorts.  I have an IO pin set to `floaterSensor = new Gpio(10, 'in', 'both');` and a watch setup to monitor changes.  Behavior works as intended for a 8 hour+ period, but then when I come back to it seems to be spamming the callback multiple times per second.  Restarting the process, behavior returns to normal.  Has this been encountered before, any ideas on what might be causing this?

fivdi, 2019-06-13:
> Has this been encountered before

I haven't seen this or had reports of it before.

> any ideas on what might be causing this?
- Perhaps the device connected to GPIO10 really is generating multiple interrupts per second. This could be verified with an oscilloscope/logic analyzer. Adding a console.log to the watch handler may provide more information too.
- Perhaps there's an issue in `store.dispatch` and not with interrupts. Maybe `store.dispatch` logs the same information multiple times in some scenarios. This could be verified by replacing `store.dispatch(...)` with `console.log(Date())`.

I'd also suggest updating to onoff@4.1.1 as onoff@2.0.0 is quite old now.

smysnk, 2019-06-13:
Sorry, thought I was running the latest version, will update.  Will try some further debugging on my side.  Thanks for the quick response.

closed 2019-06-13 by smysnk