fivdi/onoff #157: Can't install on Raspbian Buster

jhoughjr, 2019-07-14:
This appears to affect every library I've tried for node gpio on buster.
Seems I'm the only person see the issue though.

'''
pi@raspberrypi:~/fluids $ npm install onoff
npm WARN checkPermissions Missing write access to /home/pi/fluids
npm ERR! path /home/pi/fluids
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/home/pi/fluids'
npm ERR!  { [Error: EACCES: permission denied, access '/home/pi/fluids']
npm ERR!   stack:
npm ERR!    'Error: EACCES: permission denied, access \'/home/pi/fluids\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/home/pi/fluids' }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-14T21_24_15_818Z-debug.log
pi@raspberrypi:~/fluids $ 
'''

```

pi@raspberrypi:~/fluids $ sudo npm install onoff

> epoll@2.0.9 install /home/pi/fluids/node_modules/epoll
> node-gyp rebuild

gyp WARN EACCES user "pi" does not have permission to access the dev dir "/root/.cache/node-gyp/10.15.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/fluids/node_modules/epoll/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/fluids/node_modules/epoll/.node-gyp'
gyp ERR! System Linux 4.19.50-v7+
gyp ERR! command "/usr/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/fluids/node_modules/epoll
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v5.0.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@2.0.9 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@2.0.9 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-14T21_26_19_205Z-debug.log
pi@raspberrypi:~/fluids $ 

```

fivdi, 2019-07-14:
This issue isn't related to onoff. The EACCES error is occurring because user pi doesn't have the permissions required to access directory `/home/pi/fluids` or sub-directories of `/home/pi/fluids`.

Please change the permissions for `/home/pi/fluids` and it's sub-directories so that user pi has the required permissions. Don't use sudo when installing packages with npm unless it's absolutely necessary. It's not necessary for user pi to install modules with sudo in directory `/home/pi` or sub-directories of `/home/pi`, everything should work without sudo. See [here](https://github.com/fivdi/onoff/issues/90#issuecomment-376278048) for further info.

Alternatively, create a new directory, for example, `/home/pi/fluids2`, and install everything that needs to be installed in `/home/pi/fluids2` without using sudo. Then everything should work.

fivdi, 2019-07-16:
I'm going to go ahead and close this issue as it's not an onoff issue.

closed 2019-07-16 by fivdi