fivdi/onoff #158: Function watch is called erroneously

hermosillaeveris, 2019-07-16:
Example:

const Gpio = require('onoff').Gpio;

const light = new Gpio(4, 'high'); // , 'none', { activeLow: true });
const interrup = new Gpio(21, 'in', 'both');

interrup.watch((error, value) => {
  console.log(`Error: ${error}`);
  console.log(`Interrupt value to Gpio21: ${value}`);
  light.writeSync(parseInt(value));
});

const stdin = process.stdin;
stdin.setEncoding('utf-8');

stdin.on('data', data => {
  data = data.replace(/\n/, '');
  light.writeSync(parseInt(data));
});

When I write about ligh (Gpio 4), function watch is called en Gpio 21.

Why that happens?  It should only be called when the value of Gpio 21 changes

fivdi, 2019-07-16:
With an LED connected to GPIO4 and nothing connected to GPIO21, the program posted above functions correctly for me on a Raspberry Pi 3 with onoff@4.1.3 and Node v12.4.0.

Here's what I see:
```
pi@raspberrypi:~/onoff $ node -v
v12.4.0
pi@raspberrypi:~/onoff $ 
pi@raspberrypi:~/onoff $ npm list
/home/pi/onoff
└─┬ onoff@4.1.3
  ├─┬ epoll@2.0.9
  │ ├─┬ bindings@1.5.0
  │ │ └── file-uri-to-path@1.0.0
  │ └── nan@2.14.0
  └── lodash.debounce@4.0.8

pi@raspberrypi:~/onoff $ 
pi@raspberrypi:~/onoff $ cat t.js 
const Gpio = require('onoff').Gpio;

const light = new Gpio(4, 'high'); // , 'none', { activeLow: true });
const interrup = new Gpio(21, 'in', 'both');

interrup.watch((error, value) => {
  console.log(`Error: ${error}`);
  console.log(`Interrupt value to Gpio21: ${value}`);
  light.writeSync(parseInt(value));
});

const stdin = process.stdin;
stdin.setEncoding('utf-8');

stdin.on('data', data => {
  light.writeSync(parseInt(data.trim()));
});

pi@raspberrypi:~/onoff $ 
pi@raspberrypi:~/onoff $ node t.js 
0
1
0
1
0
1
^C
pi@raspberrypi:~/onoff $ 
```
Entering 0 turns the LED off, entering 1 turns the LED on.

Have you verified that GPIO4 is not shorted to GPIO21?
Is there hardware connected to GPIO21 that can change the state of GPIO21 when GPIO4 changes state?
Can you post a picture of the circuit that clearly shows what's connected to GPIO4 and GPIO21?

This doesn't appear to be an issue with onoff.

hermosillaeveris, 2019-07-17:
The environment information:

`
$ node --version
v10.16.0
$ npm list | grep onoff
└─┬ onoff@4.1.3
$uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
`
 I tried another similar configuration, in case the problem was with GPIO 4, but the same thing happens:

`
const Gpio = require('onoff').Gpio;

const light = new Gpio(20, 'high');
const interrup = new Gpio(17, 'in', 'both');

console.log('Gpio 17 mode IN:both');

interrup.watch((error, value) => {
   console.log(`Error: ${error}`);
   console.log(`Interrupt value to Gpio 17: ${value}`);
 });

const stdin = process.stdin;
stdin.setEncoding('utf-8');

let value = 0;
stdin.on('data', data => {
  console.log(`Write value ${value}`);
  light.writeSync(value);
  value = value === 0 ? 1 : 0;
});
`

That does not always happen, and it seems random. Sometimes it takes some time for it to happen:

`
Write value 0

Write value 1

Write value 0

Write value 1

Write value 0

Write value 1

Write value 0

Write value 1

Write value 0

Write value 1

Write value 0

Write value 1
Error: null
Interrupt value to Gpio 17: 1
`

hermosillaeveris, 2019-07-17:
I hope this illustrates the example:

[image]

Currently I solve the problem by caching the "previous" value of GPIO 17. When the "watch" function is executed, I check if the previous value is different from the new one to execute the corresponding actions.

fivdi, 2019-07-17:
One side of the switch is connected to GPIO17 and the other side of the switch is connected to 5V. GPIOs on the Raspberry Pi are not 5V tolerant and connecting them to 5V will damage them. The switch should be connected to 3.3V and not 5V.

fivdi, 2019-07-19:
I'll go ahead and close this issue as it's not an onoff issue

closed 2019-07-19 by fivdi