fivdi/onoff #159: Enum will help with problems

cesarjhony, 2019-08-02:
I dispend many of time searching bugs in my code, and when I found was a problem like:
pin.writeSync(true); //not work
pin.writeSync(1);     //work fine
pin.writeSync(Gpio.HIGH);     //work fine, but ENUM is not a Standard in your lib.
But, I and the most of devs use autocomplete, like a VS Code. With the STATIC or ENUM constants might save the day, the assistents find there and show us the options.
Please add all options like a static.
Your framework is amazing, but it still may be better.

fivdi, 2019-08-11:
@cesarjhony Thank you for suggesting this improvement. I'll take a look to see what can be done to improve things.

johntalton, 2020-08-03:
Ive used HIGH / LOW in my work, as exported here:

https://github.com/fivdi/onoff/blob/07d36e2407ec4b163302c67f00bcd3294006328b/onoff.js#L343

while the TS version does something a bit different:

https://github.com/fivdi/onoff/blob/07d36e2407ec4b163302c67f00bcd3294006328b/onoff.d.ts#L1

I would suggest that the js->d.ts is a stub solution at this point and there the `BinaryValue` type exists.

Anything to aid in the correctness / clearness / ease of use would be interesting as many libraries "export" this type of `const FOO = x` / `Klazz.FOO = FOO` style pattern.   @cesarjhony Could you expand on your desired Use Case?

cesarjhony, 2020-08-14:
My usage was just:
HIGH, LOW, IN, OUT;

I don't know if will better to be grouped style like:
Gpio.Level={ HIGH, LOW };
Gpio.Direction={ OUT, IN };

Thank you very much for appreciate my opinion 😊, but I have no better ideas.