fivdi/onoff #161: "EACCES: permission denied, open '/sys/class/gpio/gpio19/value'" on npm start

bmwertman, 2019-10-22:
I have a program that I wrote and successfully ran on a Raspberry Pi 2 B v1.1 I'm now trying to port that over to a Banana Pi M64 for the additional RAM so I don't have to throttle Node's memory usage.

When I start my program I get the error;
`EACCES: permission denied, open '/sys/class/gpio/gpio19/value'
```
const raspi = require('raspi') // https://github.com/nebrius/raspi
const Gpio = require('onoff').Gpio // https://github.com/fivdi/onoff+
const opticSensorA = new Gpio(19, 'in', 'both', { debounceTimeout: 200 })
```
I'm fairly certain the error is originating from the opticSensorA constant declaration.
Full error message:

OS: Debian 9(Stretch)
Node version: 8.16.1
Board: http://www.banana-pi.org/m64.html

fivdi, 2019-10-22:
The program ran successfully on the Raspberry Pi because the user that ran the program (presumably user `pi`) had the permissions required to access GPIOs using the sysfs files located at `/sys/class/gpio`. The user running the program on the Banana Pi doesn't have the required permissions. This results in EACCES errors indicating that permission is denied on the Banana Pi.

This isn't actually an onoff issue, it's a Linux permissions issue.

There are two options here.

1. Use `sudo` on the Banana Pi to to run the node program with root privileges. 
2. Give the user that wants to run the node program that accesses the GPIOs the permissions required to access those GPIOs. I'm not familiar with the Banana Pi and don't know what needs to be done to achieve this.

bmwertman, 2019-10-22:
Thanks @fivdi that's what I've found as I've continued to investigate today. 

It seems some have been able to add a file to their `udev` directory that has a `.rules` extension and calls `chmod` and `chown` to allow the access to the gpio.

I tried that but I'm not sure of how to name the `.rules` file to get Linux to see it.

For me running `sudo npm start` fails with `sudo: npm: command not found`.

fivdi, 2019-10-22:
> I tried that but I'm not sure of how to name the .rules file to get Linux to see it.

On the Raspberry Pi the file is called `/etc/udev/rules.d/99-com.rules` but to the best of my knowledge it doesn't matter what the file is called.

> For me running `sudo npm start` fails with `sudo: npm: command not found`.

does the following work:
```
sudo $(which npm) start
```
or does this work:
```
sudo $(which node) your-app-name.js
```

bmwertman, 2019-10-22:
`99-com.rules` is what I named my file, that is the path that it is at and the one liner I have there is;
```
SUBSYSTEM=="gpio", PROGRAM="/bin/sh -c 'find -L /sys/class/gpio/ -maxdepth 2 -exec chown root:gpio {} \; exec chmod 770 {} \; || true'"
```
running the last two sudo commands you just gave both resulted in;

`/usr/bin/env: 'node': No such file or directory`

fivdi, 2019-10-22:
As mentioned above, I'm afraid I don't know what needs to be done on a Banana Pi to give a normal user access to GPIOs.

>running the last two sudo commands you just gave both resulted in;
>
>/usr/bin/env: 'node': No such file or directory

It looks like node is not installed in a way that makes it usable for user `root`.

bmwertman, 2019-10-22:
This [SO post answer](https://stackoverflow.com/a/38734497/2041665) seemed to fix the Nodejs issue and I'm getting some different errors now that tell me that the sudo commands seem to have fixed the permission issue I was having. 

Not ideal but it at least gets me moving forward until I can get a more permanent fix in place. 

Thanks for the help!

closed 2019-10-22 by bmwertman

fivdi, 2019-10-22:
You're welcome. Have fun with onoff.