fivdi/onoff #162: Add support to get new Gpio without set Direction

cesarjhony, 2019-11-17:
When I get new Gpio( 13, DIRECTION ), I need to put the direction, but sometimes, the Gpio was in use of another framework, like a wiringPi ("gpio" in bash), it's not export pins. If I need to check if pin is "in" or "out", I need to do   
```js
var led = new Gpio(13);
led.direction();
```
But is result a error:   
```
{ Error: EINVAL: invalid argument, write
    at Object.writeSync (fs.js:569:3)
    at Object.writeFileSync (fs.js:1200:26)
    at Gpio.setDirection (/home/node/app/node_modules/onoff/onoff.js:277:8)
    at configureGpio (/home/node/app/node_modules/onoff/onoff.js:106:12)
    at new Gpio (/home/node/app/node_modules/onoff/onoff.js:178:5) errno: -22, syscall: 'write', code: 'EINVAL' }
```
Or error like above, about the `setDiretion("out") ` :   
```
ENOENT: no such file or directory, open '/sys/class/gpio/gpio13/direction' Error: ENOENT: no such file or directory, open '/sys/class/gpio/gpio13/direction'
    at Object.openSync (fs.js:439:3)
    at Object.writeFileSync (fs.js:1190:35)
    at Gpio.setDirection (/home/node/app/node_modules/onoff/onoff.js:277:8)
    at DIRECT_GPIO_PinController.setModeAsOutput (/home/node/app/src/utils/pinControllers/HardControllers/DIRECT_GPIO_PinController.js:34:22)
    at Promise (/home/node/app/src/utils/pinControllers/WritePinController.js:128:28)
    at new Promise (<anonymous>)
    at Object.pulsoPromise [as pulseP] (/home/node/app/src/utils/pinControllers/WritePinController.js:127:28)
    at model.Carga.pulseP (/home/node/app/src/models/carga.js:48:23)
    at pulsarCarga (/home/node/app/src/controllers/api/carga/comando/pulse.js:33:21) meta= undefined
```
   
One workaroud that works is:
```js
try {
    this.gpioPin = new Gpio( this._pin ); //bug no gpio se iniciar sem a "direction" in/out
} catch (error) {
    this.gpioPin = new Gpio( this._pin );
}
```
The regular use `new Gpio( this._pin, DIRECTION );` is fine, but is not work for my logic (check direction before...). If onoff have another way to do this check, tell me please.
My enviromniment used is:   
Docker, inside a real raspberry pi zero w, runing as sudo and `... --privileged ...` (meaning full access to devices);   
Node 10;   
onoff 5.0.0;

fivdi, 2019-11-17:
Internally [onoff uses sysfs files](https://github.com/fivdi/onoff#how-does-onoff-work) located at `/sys/class/gpio` to access GPIOs. This makes it possible to use onoff on many platforms, not only the Raspberry Pi.

When platform specific tools like the Wiring Pi gpio utility are used to configure GPIOs using platform specific mechanisms such as writing directly to registers, it's not possible for onoff to determine how the GPIOs have been configured. Attempting to do this would make onoff platform specific rather than platform independent as is currently the case.

An example of a call to the Wiring Pi gpio utility that uses Raspberry Pi specific registers is:
```
gpio -g mode 17 out
```
However, the Wiring Pi gpio utility also supports exporting pins using sysfs files, for example:
```
gpio export 17 out
```
If the Wiring Pi gpio utility exported pins using sysfs files, it would be possible to extend onoff to do what you would like to do.

That being said, I'm not sure if I want to extend onoff to make this possible. The use case mentioned above is:
```js
var led = new Gpio(13);
led.direction();
```
In this use case it's almost certainly the case that the GPIO should be configured as an output. To be honest, I'm having difficulties thinking of a use case where a program should use GPIO 13 without knowing in advance whether it's an input or an output.

Do you have a valid use case for not knowing whether a GPIO that your program must use is an input or output? If so, what is that use case?

I do actually have another Raspberry Pi specific library called [pigpio](https://github.com/fivdi/pigpio) that may be more suitable for your use case. It allows Gpio objects to be created without specifying their mode. For an example of doing this take a look at the [gpio-info.js](https://github.com/fivdi/pigpio/blob/master/example/gpio-info.js) which prints the mode and value of all GPIOs on a Raspberry Pi.

cesarjhony, 2019-11-18:
Thank you for so care message. I think that use case is often, because I did run in onoff older versions without any problems, but directly in raspberry, without Docker.

closed 2019-11-18 by cesarjhony