fivdi/onoff #163: Add support for remote GPIO

Sawtaytoes, 2019-12-04:
Raspberry Pi supports Remote GPIO.

[image]

It'd be great if I could run `onoff` on my development machine and pass in the hostname of the Raspberry Pi device to remotely access the values on the GPIO pins without having to test my software on the machine itself.

fivdi, 2019-12-05:
The "Remote GPIO" option in the screenshot above is only related to GPIO Zero and pigpio. It's not a general option that will activate remote GPIO for all tools and libraries. As such, it will not enable remote GPIO for onoff.

See https://gpiozero.readthedocs.io/en/stable/remote_gpio.html

Remote GPIO is outside the scope of onoff. You could however implement a server on the Pi which controls GPIOs on the Pi and then communicate with the server on the Pi from your development machine.

closed 2019-12-05 by fivdi

Sawtaytoes, 2019-12-05:
About exposing GPIO myself, that's what I've ended up doing. I was hoping to reduce the custom software load on my remote Pis, but I guess that's not possible.

Thanks for going into detail about Remote GPIO. Documentation on it is confusing for someone who's not in-the-know. Even seeing your response. I have no clue what GPIO Zero and pigpio are or how they're related, but after looking them up, it's clear Remote GPIO is designed for a completely different use case to what onoff is solving.