fivdi/onoff #164: Blinking led not working

kevinpallado, 2019-12-31:
After following the tutorials [here](https://www.w3schools.com/nodejs/nodejs_raspberrypi_blinking_led.asp) the led wont blink. I am using rpi3 model b+ running on a raspbian lite os. Does this package still works? Or maybe I do something wrong

fivdi, 2019-12-31:
>Does this package still works?

Yes, this package still works and the program to blink an LED on the page linked to above should also work correctly.

- Have you verified that the LED is connected up correctly?
- Are you sure that the LED is connected the right way around?
- Is there any other software running that could be using GPIO4 thus preventing the LED blink program from running?

kiwdahc, 2019-12-31:
I can verify that it works just did this pages guide today. What I can say though is that npm install was failing with node version 10. When I moved back to 8 it started working, also verified it in a docker file and build constantly failed until reverting to node 8. It appears epoll only works on 8 which is a dependency of this so you may need to update the readme if I am not mistaken.

fivdi, 2019-12-31:
@kiwdahc Thank your for confirming that it works. I just installed onoff with Node.js v8, v10, v12, v13 and v14-nightly. As can be see below onoff installed and ran successfully with all versions of Node.js.

Can you post the complete output of `npm install onoff` when attempting to install with Node.js v10 please as all looks ok on my side?

### Node.js v8.17.0
```
pi@raspberrypi:~/onoff $ rm -rf *
pi@raspberrypi:~/onoff $ node -v
v8.17.0
pi@raspberrypi:~/onoff $ npm i onoff

> epoll@3.0.0 install /home/pi/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 10.13s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

### Node.js v10.18.0
```
pi@raspberrypi:~/onoff $ rm -rf *
pi@raspberrypi:~/onoff $ node -v
v10.18.0
pi@raspberrypi:~/onoff $ npm i onoff

> epoll@3.0.0 install /home/pi/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 8.802s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

[... 79 lines cut ...]
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.
npm WARN You are using a pre-release version of node and things may not work as expected

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 6.25s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

kevinpallado, 2019-12-31:
Hi @fivdi  thanks for the quick response, I'll try the mentioned solutions above. I'll back after I test this

kevinpallado, 2019-12-31:
@fivdi this is my code on trying to turn on/off the gpio

`const gpio = require('onoff').Gpio;
const  waterpump = new gpio(19, 'out');
const lightcontrol = new gpio(26, 'out');

s.clients.forEach(function (client) { //broadcast incoming message to all clients (s.clients)1
            if (client.isAlive === false) return client.terminate();
            ws.isAlive = false;
            ws.ping(noop);

            var msg_parse = JSON.parse(message);
            client_data = [];
            lightLogControls(msg_parse.lightI);
            soilLogControls(msg_parse.soilM);
            // msg_parse.soilM < 150 ? waterpump.writeSync(1) : waterpump.writeSync(0);
            // msg_parse.lightI < 30 ? lightcontrol.writeSync(1) : lightcontrol.writeSync(0);

            for(x=0; x < client_connect; x++)
            {
                client_data.push({ tempc: msg_parse.tempC, humidity: msg_parse.humid, soilm: msg_parse.soilM, lightInt: msg_parse.lightI });
            }
            console.log(client_connect);
            });`

fivdi, 2019-12-31:
@kevinpallado Here is the original issue that you mentioned in the first post above:

> After following the tutorials [here](https://www.w3schools.com/nodejs/nodejs_raspberrypi_blinking_led.asp) the led wont blink.

My assumption was that the LED blink program on the page linked to in the first post doesn't work. I now see that this doesn't appear to be the case and the issue is something very different.

The code now posted [here](https://github.com/fivdi/onoff/issues/164#issuecomment-569913388) isn't remotely related to the original issue. Also, the code now posted isn't a complete program and contains very little code related to onoff. I'm sorry but I can debug your code. This is something that you'll have to do yourself.

I'm going to go ahead and close this issue. If it turns out that the issue is related to onoff the issue can be reopened.

closed 2019-12-31 by fivdi

kevinpallado, 2020-01-02:
Hi @fivdi It was my mistake to post my own code, anyways I tried using this code shown below this is based on the tutorials, after running the command the led wont blink. I did test the LED if it is working and it did. I use python code to test the led, did I miss something?

```
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var LED = new Gpio(4, 'out'); //use GPIO pin 4, and specify that it is output
var blinkInterval = setInterval(blinkLED, 250); //run the blinkLED function every 250ms

function blinkLED() { //function to start blinking
  if (LED.readSync() === 0) { //check the pin state, if the state is 0 (or off)
    LED.writeSync(1); //set pin state to 1 (turn LED on)
  } else {
    LED.writeSync(0); //set pin state to 0 (turn LED off)
  }
}

function endBlink() { //function to stop blinking
  clearInterval(blinkInterval); // Stop blink intervals
  LED.writeSync(0); // Turn LED off
  LED.unexport(); // Unexport GPIO to free resources
}

setTimeout(endBlink, 5000); //stop blinking after 5 seconds
```

kevinpallado, 2020-01-02:
I already know the problem, probably the pin number confusion is the cause.

fivdi, 2020-01-02:
GPIO4 is on pin 7 of the Raspberry Pi GPIO header.

See https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/raspberry_pi_3_model_b_plus_gpio.jpg

Is the LED connected to pin 7?