fivdi/onoff #165: Npm install failure when installing from node10

kiwdahc, 2019-12-31:
I am not sure of the exact issue causing this or if it was caused on my end but here is the call stack when trying to `npm ci` from node v10 in a docker file.

**Callstack**
```bash
> epoll@3.0.0 install /usr/src/app/node_modules/epoll
> node-gyp rebuild

qemu: Unsupported syscall: 397
qemu: Unsupported syscall: 397
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.9.125-linuxkit
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/src/app/node_modules/epoll
gyp ERR! node -v v10.18.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@3.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-12-31T21_45_45_749Z-debug.log
```

**Dockerfile**
```bash
FROM arm32v7/node:10.18.0-buster

ARG APP_DIR=/usr/src/app/

RUN mkdir -p $APP_DIR
WORKDIR $APP_DIR

RUN mkdir -p build dist logs
RUN chmod 777 build dist logs .

COPY package.json package-lock.json $APP_DIR

COPY . .

RUN npm ci && npm cache clean --force
```

If you change the base image to `8.17.0-buster` the build works.

The only package in the package.json is.
```json
"dependencies": {
    "onoff": "^5.0.1"
}
```

fivdi, 2020-01-01:
I'm not really sure what the problem is here but two things that stick out in the output provided above are:
```
qemu: Unsupported syscall: 397
qemu: Unsupported syscall: 397
```
and:
```
gyp ERR! System Linux 4.9.125-linuxkit
```
As can be seen [here](https://github.com/nodejs/node/blob/v10.18.0/deps/uv/src/unix/linux-syscalls.c#L195-L198), syscall 397 is `statx` and as can be seen in the "VERSIONS"  section of the [statx man-page](http://man7.org/linux/man-pages/man2/statx.2.html) `statx` was added to Linux in kernel 4.11. On the other hand, the output provided above indicates that Linux kernel 4.9.125 is being used.

Debian 9 (Stretch) comes with a Linux kernel version between 3.16 and 4.9. Given that "System Linux 4.9.125-linuxkit" which is presumably Debian 9 is being used, I would imagine that the dockerfile should be using `arm32v7/node:10.18.0-stretch` rather than `arm32v7/node:10.18.0-buster`.

Please try with `10.18.0-stretch` to see if it works.

fivdi, 2020-01-01:
As can be seen below, onoff installs and runs without problems on Raspbian Buster with Node.js v8, v10, v12, v13 and v14-nightly.

### Node.js v8.17.0
```
pi@raspberrypi:~/onoff $ rm -rf *
pi@raspberrypi:~/onoff $ node -v
v8.17.0
pi@raspberrypi:~/onoff $ npm i onoff

> epoll@3.0.0 install /home/pi/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 10.13s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

### Node.js v10.18.0
```
pi@raspberrypi:~/onoff $ rm -rf *
pi@raspberrypi:~/onoff $ node -v
v10.18.0
pi@raspberrypi:~/onoff $ npm i onoff

> epoll@3.0.0 install /home/pi/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 8.802s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

### Node.js v12.14.0
```
[... 77 lines cut ...]
> node-gyp rebuild

make: Entering directory '/home/pi/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/onoff/node_modules/epoll/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/onoff/package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.
npm WARN You are using a pre-release version of node and things may not work as expected

+ onoff@5.0.1
added 6 packages from 12 contributors and audited 6 packages in 6.25s
found 0 vulnerabilities

pi@raspberrypi:~/onoff $ node node_modules/onoff/examples/wait-for-interrupt.js 
Please press the button on GPIO4...
Button pressed!, its value was 1
pi@raspberrypi:~/onoff $ 
```

fivdi, 2020-01-12:
@kiwdahc Was it possible to make any progress here?

fivdi, 2020-01-24:
I'm going to go ahead and close this issue as it's highly unlikely to be an onoff issue. If it does turn out to be an onoff issue the issue can be reopened.

closed 2020-01-24 by fivdi