fivdi/onoff #166: LED goes out without specifying a 0 value in Node JS

sts-ryan-holton, 2020-01-18:
Having some trouble with LEDs on a Raspberry Pi 4, using the Jam HAT. Apparently, turning a LED on via a Node JS script will then cause the LED to go off straight after my `writeSync()` even though there's nothing else specifying it to be turned off, why?I want the LED to remain on until I specify `0`

```javascript
const Gpio = require('onoff').Gpio
const led = {
  redOne: new Gpio(6, 'out')
}

led.redOne.writeSync(1)
```

fivdi, 2020-01-19:
To the best of my knowledge the program posted above should turn the LED on. The program should then terminate as it has no remaining work to do and the LED should remain on.

If the program shown above turns the LED off, does the following program turn the LED on and terminate leaving the LED on? If so, a value of 0 needs to be written to the GPIO to turn the LED on.
```js
const Gpio = require('onoff').Gpio
const led = {
  redOne: new Gpio(6, 'out')
}

led.redOne.writeSync(0)
```
Does the following program blink the LED?
```js
const Gpio = require('onoff').Gpio
const led = {
  redOne: new Gpio(6, 'out')
}

setInterval(() => {
  led.redOne.writeSync(led.redOne.readSync() ^ 1)
}, 500)
```
I'm assuming that there is no other software running on the Raspberry Pi that is assessing GPIO 6. Is this assumption correct?

fivdi, 2020-01-24:
@sts-ryan-holton Was it possible to make any progress here?

closed 2020-01-26 by sts-ryan-holton

nkemcels, 2020-10-09:
I'm having this same issue on the raspberry Pi 4. 
@sts-ryan-holton did you find a solution to this?

fivdi, 2020-10-10:
@nkemcels I would suggest opening a new issue that describes the problem. The new issue should include a complete but small program that can be used to reproduce the problem. It should also include a photo of the circuit where the wires connecting the LED to the Raspberry Pi 4 are visible.