fivdi/onoff #168: How to read power status

emmtte, 2020-02-12:
Hi, thanks for this great npm module.
It's help me a lot. 
But just a question I need to use IRQ to read power down with vcgencmd
Is your npm have implemented vcgencmd IRQ?

https://raspberrypi.stackexchange.com/questions/60593/how-raspbian-detects-under-voltage

fivdi, 2020-02-12:
I'm afraid this isn't possible with onoff.

onoff can be used to detect interrupts on GPIO inputs that are accessable via the sysfs files located at `/sys/class/gpio`. The red power LED is connected to GPIO35 but GPIO35 can not be accessed with the sysfs files located at `/sys/class/gpio`.

It may be possible to achieve what you would like to achieve with `fs.watch`, i.e., something like this:

```js
const fs = require('fs');

fs.watch('/sys/class/leds/led1/brightness', (eventType, filename) => {
  fs.readFile('/sys/class/leds/led1/brightness', (err, data) => {
    console.log(data);
  });
});
```

emmtte, 2020-02-12:
Great, thank you very much for your help.
I will test your code as soon as possible

closed 2020-02-12 by emmtte