fivdi/onoff #172: npm i failing in windows

shekhars271991, 2020-05-28:
> epoll@3.0.0 install D:\AI\SI\Agent\agentServer\node_modules\epoll
> node-gyp rebuild

D:\AI\SI\Agent\agentServer\node_modules\epoll>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Command failed: C:\Users\KH2554\AppData\Local\Programs\Python\Python37\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:294:12)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\AI\SI\Agent\agentServer\node_modules\epoll
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN bd_orders@1.0.0 No description
npm WARN bd_orders@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@3.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\KH2554\AppData\Roaming\npm-cache\_logs\2020-05-28T09_33_02_183Z-debug.log

fivdi, 2020-05-28:
`npm i onoff` working on Windows:

```
c:\src\onoff>node -v
v10.15.3

c:\src\onoff>npm i onoff

> epoll@4.0.0 install c:\src\onoff\node_modules\epoll
> node-gyp rebuild

c:\src\onoff\node_modules\epoll>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modu
les\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\
node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  win_delay_load_hook.cc
  Generating code
  All 1 functions were compiled because no usable IPDB/IOBJ from previous compilation was found.
  Finished generating code
  epoll.vcxproj -> c:\src\onoff\node_modules\epoll\build\Release\\epoll.node
  epoll.vcxproj -> c:\src\onoff\node_modules\epoll\build\Release\epoll.pdb (Full PDB)
npm WARN test@1.0.0 No description
npm WARN test@1.0.0 No repository field.

+ onoff@6.0.0
added 6 packages from 12 contributors and audited 6 packages in 4.525s
found 0 vulnerabilities

c:\src\onoff>
```

`npm i epoll@3.0.0` working on Windows:

```
c:\src\onoff>npm i epoll@3.0.0

> epoll@3.0.0 install c:\src\onoff\node_modules\epoll
> node-gyp rebuild

c:\src\onoff\node_modules\epoll>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modu
les\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\
node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  win_delay_load_hook.cc
  Generating code
  All 1 functions were compiled because no usable IPDB/IOBJ from previous compilation was found.
  Finished generating code
  epoll.vcxproj -> c:\src\onoff\node_modules\epoll\build\Release\\epoll.node
  epoll.vcxproj -> c:\src\onoff\node_modules\epoll\build\Release\epoll.pdb (Full PDB)
npm WARN saveError ENOENT: no such file or directory, open 'c:\src\onoff\package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open 'c:\src\onoff\package.json'
npm WARN onoff No description
npm WARN onoff No repository field.
npm WARN onoff No README data
npm WARN onoff No license field.

+ epoll@3.0.0
added 4 packages from 9 contributors and audited 4 packages in 4.979s
found 0 vulnerabilities

c:\src\onoff>
```
It looks like something isn't setup correctly on your system.

fivdi, 2020-05-28:
Perhaps it's the version of Python being used. I'm using Python v2.7.16.

shekhars271991, 2020-05-28:
Do we need any python version installed? why?

shekhars271991, 2020-05-28:
I install python 2.7.18 (it sees node-gyp is not supported in python 3). But the install still failed with the following tracelog.

` npm i onoff

> epoll@4.0.0 install D:\AI\SI\Agent\agentServer\node_modules\epoll
> node-gyp rebuild

D:\AI\SI\Agent\agentServer\node_modules\epoll>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "4.0".
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\AI\SI\Agent\agentServer\node_modules\epoll
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN bd_orders@1.0.0 No description
npm WARN bd_orders@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@4.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@4.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\KH2554\AppData\Roaming\npm-cache\_logs\2020-05-28T15_20_20_363Z-debug.log`

fivdi, 2020-05-28:
> Do we need any python version installed?

Yes

> why?

Because onoff depends on epoll and epoll is a native module that is implemented in C++. When native modules are installed they are built by node-gyp which in turn requires Python.

A Visual C++ compiler is also needed.

A potential alternative to avoid these issues is to add onoff to the optional dependencies in package.json. Then error messages will be displayed but the installation process will continue. onoff will not be of any usage on a Windows machine.

shekhars271991, 2020-05-28:
Thanks, It helped. :)

closed 2020-05-28 by shekhars271991