fivdi/onoff #173: Writing serveal gpios

buildog, 2020-06-04:
Using Waveshare relay board (https://www.waveshare.com/wiki/RPi_Relay_Board_(B)) with a RPI 3B 

```js
import { Gpio } from 'onoff';

const gpios = [
  { id: 26, state: 0 },
  { id: 21, state: 0 },
  { id: 20, state: 0 },
  { id: 19, state: 0 },
  { id: 16, state: 0 },
  { id: 13, state: 0 },
  { id: 6, state: 0 },
  { id: 5, state: 0 }
]

// Toogle off all gpios
gpios.map((gp) => { new Gpio(gp.id, 'out').write(1) })

// Ref up/down GPIOS
const up = new Gpio(26, 'out');
const down = new Gpio(21, 'out');

up.write(0);
```

This script write both up and down while removing `const down = new Gpio(21, 'out');` only write the pin 26

buildog, 2020-06-04:
Wrapping the new Gpio in a function fixed the issue

closed 2020-06-04 by buildog

fivdi, 2020-06-04:
I'd recommend creating one Gpio object per GPIO rather than two Gpio objects per GPIO. The code posted creates two Gpio objects for GPIO26 and two for GPIO21.

Also, please read the complete documentation for the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options).