fivdi/onoff #174: UncaughtException: EPERM: operation not permitted

rajanlagah, 2020-07-30:
I am using 

```

var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var liveA = new Gpio(21, 'out');
var liveB = new Gpio(26, 'out');
var liveC = new Gpio(19, 'out');
var liveD = new Gpio(13, 'out');
var relay1 = new Gpio(16, 'out');
```

then code is working fine but when i 

var relay2 = new Gpio(20, 'out'); 
and do 
```

function a (){
        relay2.writeSync(1);
}
a();
```

It is giving me error 

```
{ error:
   { Error: EPERM: operation not permitted, write
  at Object.writeSync (fs.js:573:3)
        at Gpio.writeSync (/home/../node_modules/onoff/onoff.js:243:8)
         at Timeout.blinkLED [as _onTimeout] (/home/..../utilities.js:87:16)
        at ontimeout (timers.js:436:11)
       at tryOnTimeout (timers.js:300:5)
        at listOnTimeout (timers.js:263:5)
        at Timer.processTimers (timers.js:223:10) errno: -1, syscall: 'write', code: 'EPERM' },
   level: 'error',

```

Any idea whats going wrong ?

fivdi, 2020-07-30:
What type of system is the error occurring on? A Raspberry Pi 0, 1, 2, 3 or 4? A BeagleBone? Something else?

If a Raspberry Pi with a 40-pin GPIO header is being used please ensure that the SPI1 peripheral is not activated for usage as activating SPI1 will reserve GPIO20 for SPI thus making it unusable for other purposes. 

Please post a complete (but short) program that can be run and used to reproduce the error rather than snippets of code.

Please post the content of `/boot/config.txt`

fivdi, 2020-07-30:
If GPIO20 is configured as an output and later configured as an input, calling `writeSync` for GPIO20 will result in an `EPERM` error. This is to be expected as it's not permitted to call `writeSync` for a GPIO configured as an input.

For example, running this program:
```js
const Gpio = require('onoff').Gpio;

const relay2 = new Gpio(20, 'out');
const input = new Gpio(20, 'in');

function a() {
  relay2.writeSync(1);
}

a();
```
results in this `EPERM` error:
```
internal/fs/utils.js:259
    throw err;
    ^

Error: EPERM: operation not permitted, write
    at Object.writeSync (fs.js:694:3)
    at Gpio.writeSync (/home/pi/onoff/node_modules/onoff/onoff.js:243:8)
    at a (/home/pi/onoff/t.js:7:10)
    at Object.<anonymous> (/home/pi/onoff/t.js:10:1)
    at Module._compile (internal/modules/cjs/loader.js:1185:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1205:10)
    at Module.load (internal/modules/cjs/loader.js:1034:32)
    at Function.Module._load (internal/modules/cjs/loader.js:923:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47 {
  errno: -1,
  syscall: 'write',
  code: 'EPERM'
}
```

fivdi, 2020-08-03:
@rajanlagah Any news on this one?

rajanlagah, 2020-08-04:
I do have 
`var relay2 = new Gpio(20, 'out'); 
`and then 
```
function enable( ){
    if(stage == 1){
        relay2.writeSync(1);
    }
    if(stage == 2){
        liveD.writeSync(1);
    }
    if(stage == 3){
        relay2.writeSync(0);
    }
    if(stage == 4){
        liveD.writeSync(1);
    }
    if(stage < 4){
        setTimeout(enable(stage+1),500)
    }
}
```

fivdi, 2020-08-04:
As mentioned above, please post a complete (but short) program that can be run and used to reproduce the error rather than snippets of code.

I'm afraid I can't provide any help without a complete (but short) program that can be used to reproduce the error.

rajanlagah, 2020-08-04:
```
const LCD = require('lcdi2c');
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var liveA = new Gpio(21, 'out');
var liveB = new Gpio(26, 'out');
var liveC = new Gpio(19, 'out');
var liveD = new Gpio(13, 'out');
var relay1 = new Gpio(16, 'out');
var relay2 = new Gpio(20, 'out'); 

function enable( ){
    if(stage == 1){
        relay2.writeSync(1);
    }
    if(stage == 2){
        liveD.writeSync(1);
    }
    if(stage == 3){
        relay2.writeSync(0);
    }
    if(stage == 4){
        liveD.writeSync(1);
    }
    if(stage < 4){
        setTimeout(enable(stage+1),500)
    }
}

enable()
```

This is my short code

fivdi, 2020-08-04:
What is the output of that program when you run it?
Were is stage defined?

rajanlagah, 2020-08-04:
```

var blinkInterval = "" // setInterval(blinkLED, 500); //run the blinkLED function every 250ms
var stepNumber = 0;

function blinkLED() { //function to start blinking
  if (stepNumber == 0) {

        relay2.writeSync(1);
    vending = true;
    console.log("Turned off card");
    stepNumber = 1;

  } else if(stepNumber == 1){

    relay2.writeSync(0);
    stepNumber = 2;
    console.log("Turned on card");
    endBlink();
  }
}

function endBlink() { //function to stop blinking

  clearInterval(blinkInterval); // Stop blink intervals

  relay2.writeSync(0);
  console.log("Completed routine");
}

setTimeout(endBlink, 1000*60*5);

function enable( ){
    blinkInterval = setInterval(blinkLED, 500);
}
enable()
```

Another way of doing that but having same issue

rajanlagah, 2020-08-04:
> What is the output of that program when you run it?
> Were is stage defined?

Yes yes it was a global variable

fivdi, 2020-08-04:
What is the output of [this](https://github.com/fivdi/onoff/issues/174#issuecomment-668415410) program when you run it?

rajanlagah, 2020-08-04:
```
internal/fs/utils.js:259
    throw err;
    ^

Error: EPERM: operation not permitted, write
    at Object.writeSync (fs.js:694:3)
    at Gpio.writeSync (/home/pi/onoff/node_modules/onoff/onoff.js:243:8)
    at a (/home/pi/onoff/t.js:7:10)
    at Object.<anonymous> (/home/pi/onoff/t.js:10:1)
    at Module._compile (internal/modules/cjs/loader.js:1185:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1205:10)
    at Module.load (internal/modules/cjs/loader.js:1034:32)
    at Function.Module._load (internal/modules/cjs/loader.js:923:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47 {
  errno: -1,
  syscall: 'write',
  code: 'EPERM'
}
```

THIS

fivdi, 2020-08-04:
That is not the output. That was copied that from [here](https://github.com/fivdi/onoff/issues/174#issuecomment-666603040). I'm afraid I can't provide any further help, good luck.

closed 2020-08-04 by fivdi