fivdi/onoff #179: Watch is not working on a button but readSync is

andre-lima, 2020-10-28:
I read some of the other issues here, and changed the config.txt file so the GPIO17 is a pullup type input.

`gpio=17=pu`

Although, the *watch* method is not triggered, but the readSync (inside a setInterval) works correctly no matter which GPIO input i plug my button to.

Is something else missing?

Running Node 12.19, onoff 6.0.1

andre-lima, 2020-10-28:
I just realized i had added this line to the config.txt when considering using the *rpio* package:
`dtoverlay=gpio-no-irq`

This line was preventing the watch function from being triggered.

Sorry for the trouble.

closed 2020-10-28 by andre-lima

johntalton, 2020-10-28:
@andre-lima it may be worth and addition to the documentation ... i was unaware of such a flag - any documentation may also be helpfull