fivdi/onoff #180: fs.js:114     throw err;     ^  Error: EBUSY: resource busy or locked, write

humblewizard, 2020-11-25:
I'm trying to run the example code for onoff and I'm getting this error:

pi@raspberrypi:~ $ sudo node led-blink.js
fs.js:114
    throw err;
    ^

Error: EBUSY: resource busy or locked, write
    at Object.writeSync (fs.js:573:3)
    at Object.writeFileSync (fs.js:1204:26)
    at exportGpio (/home/pi/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/node_modules/onoff/onoff.js:172:36)
    at Object.<anonymous> (/home/pi/webserver.js:14:11)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)

The node version is 10.23 running on a Raspberry Pi 4

What should I try to fix this?

fivdi, 2020-11-25:
Please post the code that results in this error and the content of file /boot/config.txt.

humblewizard, 2020-11-25:
The code is:

```
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
var LED = new Gpio(4, 'out'); //use GPIO pin 4, and specify that it is output
var blinkInterval = setInterval(blinkLED, 250); //run the blinkLED function every 250ms

function blinkLED() { //function to start blinking
  if (LED.readSync() === 0) { //check the pin state, if the state is 0 (or off)
    LED.writeSync(1); //set pin state to 1 (turn LED on)
  } else {
    LED.writeSync(0); //set pin state to 0 (turn LED off)
  }
}

function endBlink() { //function to stop blinking
  clearInterval(blinkInterval); // Stop blink intervals
  LED.writeSync(0); // Turn LED off
  LED.unexport(); // Unexport GPIO to free resources
}
```

The content of /boot/config.txt:

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

#next has been added to get the serial temp sensor interface to run
dtoverlay=w1-gpio

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

fivdi, 2020-11-25:
The following line in /boot/config.txt is most likely the cause of the problem:
```
dtoverlay=w1-gpio
```
Please remove or comment out the line, reboot the Pi and try again.

The w1-gpio overlay reserves GPIO4 for 1-Wire devices. If GPIO4 is reserved for 1-Wire, it can't be used by onoff and attempting to use it will result in EBUSY errors.

humblewizard, 2020-11-25:
Yes, that fixed my problem. I have serial temperature sensors I'm reading on pin 4 as I recall now.  Thanks :)

fivdi, 2020-11-25:
That's good news. Thanks for the feedback.

closed 2020-11-25 by fivdi