fivdi/onoff #181: Nvidia jetsons compatibiity

marcoburrometo, 2020-12-06:
Hello all,
I hope this is the right place to ask this question.
Is this library working with nvidia jetsons?

What should i do to make it work with it?

fivdi, 2020-12-06:
I'm not familiar with NVIDIA Jetson boards so I can't provide a definitive answer. If the boards support the Linux GPIO sysfs interface for userspace access to GPIOs then it should work. See also the ["How Does onoff Work"](https://github.com/fivdi/onoff#how-does-onoff-work) section of the readme.

marcoburrometo, 2020-12-06:
Thanks, i will try it soon amd let you know

fivdi, 2020-12-06:
> Thanks, i will try it soon amd let you know

Yes, please do. It world be interesting to know if it works.

fivdi, 2021-01-04:
@marcoburrometo did you have any luck with onoff and the NVIDIA Jetson board?

fivdi, 2021-01-15:
There is currently no TODO for onoff here so I'll go ahead and close the issue. If it turns out that there is an issue we can reopen the issue.

closed 2021-01-15 by fivdi