fivdi/onoff #182: Problem using watch function with relay

idmarco, 2021-01-04:
Hi!

I have an IR sensor with relay and connected to raspberry to gpio 17. In python I can see the change when I get from the relay, but I don't get it with onoff. I have tried different gpio, and different settings. I can see the status of the relay with the read function.
My code in python:
import time
import sys
 
SENSOR_PIN = 17
 
GPIO.setmode(GPIO.BCM)
GPIO.setup(SENSOR_PIN, GPIO.IN)

def test(channel):
  print('State change!')

def main():
  try:
    GPIO.add_event_detect(SENSOR_PIN , GPIO.RISING, callback=test)
    while True:
        time.sleep(1)
  except KeyboardInterrupt:
    print ("Finish...")
  GPIO.cleanup()
  
if __name__ == '__main__':
  main()

With watch function dont work:
       var self = this
       this.pirRemoteControl = new Gpio(17, 'in', 'both') // I test with 'high', 'low', ...
        
       this.pirRemoteControl .watch(function (err, value) {
          console.log('Status change');
        });

This code works:
      setInterval(function(){
          self.pirRemoteControl.read((err, value) => {
            console.log("State value:"+value)
            if (err) {
              throw err;
            }
          });
        },100);

Do you know why dont run with watch function?

Thanks for all!!!

fivdi, 2021-01-04:
Hi, please post a complete program which is as short as possible that can be used to reproduce the problem.

idmarco, 2021-01-04:
Thanks for your reply. I try to explain better.
I use an IR sensor with remote control, and I need to know if I am pressing the button.
[image]

I test the IR sensor with python, works fine:
import time
import sys
 
SENSOR_PIN = 17
GPIO.setmode(GPIO.BCM)
GPIO.setup(SENSOR_PIN, GPIO.IN)

def test(channel):
  print('State change!')

def main():
  try:
    GPIO.add_event_detect(SENSOR_PIN , GPIO.RISING, callback=test)
    while True:
        time.sleep(1)
  except KeyboardInterrupt:
    print ("Finish...")
  GPIO.cleanup()
  
if __name__ == '__main__':
  main()

In nodejs, i create a module for magicMirror.
This code works fine:

'use strict'

var NodeHelper = require('node_helper')
const Gpio = require('onoff').Gpio;

module.exports = NodeHelper.create({
  
  isLoaded: false,
  config:   null,
  
  start: function () {
    console.log(this.name + ' helper started ...')
  },
  
  socketNotificationReceived: function (notification, payload) {
    var self = this
    if (notification === 'CONFIG') {
        if (!this.isLoaded) {
            this.config = payload
            this.isLoaded = true
        }
        
        this.pirRemoteControl = new Gpio(17, 'in', 'both')
        
        setInterval(function(){
          self.pirRemoteControl.read((err, value) => {
            console.log("Status change to:"+value)
            if (err) {
              throw err;
            }
          });
        },100);
    }
  },
})

But this one, dont work;

'use strict'

var NodeHelper = require('node_helper')
const Gpio = require('onoff').Gpio;

module.exports = NodeHelper.create({
  
  isLoaded: false,
  config:   null,
  
  start: function () {
    console.log(this.name + ' helper started ...')
  },
  
  socketNotificationReceived: function (notification, payload) {
    var self = this
    if (notification === 'CONFIG') {
        if (!this.isLoaded) {
            this.config = payload
            this.isLoaded = true
        }
        
        this.pirRemoteControl = new Gpio(17, 'in', 'both')
        
        this.pirRemoteControl.watch(function (err, value) {
          if (err) {
            console.log("Error: "+err);
          }else{
            console.log("Status change: "+value)
          }
        });
    }
  },
})

Its possible thant watch function dont work with relay?

Thanks!!

fivdi, 2021-01-05:
In general, `watch` should work. I don't understand why it doesn't work here though.

If the following program is run, does it print messages when the button the remote control is pressed?
```js
'use strict'

const Gpio = require('onoff').Gpio;

let pirRemoteControl = new Gpio(17, 'in', 'both');
        
pirRemoteControl.watch(function (err, value) {
  if (err) {
    console.log("Error: "+err);
  } else {
    console.log("Status change: "+value);
  }
});
```
If you change this line from your program:
```js
        this.pirRemoteControl = new Gpio(17, 'in', 'both')
```
to this:
```js
        console.log('About to create the Gpio object...');
        this.pirRemoteControl = new Gpio(17, 'in', 'both')
```
is the message 'About to create the Gpio object...' displayed? How often is the message displayed?

idmarco, 2021-01-05:
Yes, the message display in log file.
I use the watch function too with other sensor (move sensor) and works correcty.
Tomorrow I will do some tests without the use of magicmirror.
Could the relay have something to do with it?

Thanks

fivdi, 2021-01-05:
Just so that I understand, you are saying that when the following program is run, it functions correctly?
```js
'use strict'

const Gpio = require('onoff').Gpio;

let pirRemoteControl = new Gpio(17, 'in', 'both');
        
pirRemoteControl.watch(function (err, value) {
  if (err) {
    console.log("Error: "+err);
  } else {
    console.log("Status change: "+value);
  }
});
```

idmarco, 2021-01-05:
Hi!

I have done several tests starting from 0, and so now it works correctly. I can't quite understand what the problem was, maybe because of some other bookstore. I do not know.
But now it works perfectly, thank you very much and sorry for the inconvenience

fivdi, 2021-01-05:
No Problem. Good to hear it's working now.

closed 2021-01-05 by fivdi

gwest7, 2022-05-09:
I hope I can use this thread for a related question as I don't want to log an issue if there is no need.

I `watch` a few pins setup with `"in"` and `"both"`. I have a few PIR sensors wired to these. With movement the callback value parameter is `1` and once idle the value parameter is `0`.

Whenever I write `1` to any of the output pins (with cheap relays hooked up) the `watch` callbacks of all the input pins are called with the value `0`. The relays control 10W LED lights (set up not to interfere with the PIR sensors).

I suspect this is circuit related, but even so I would expect the `watch` callbacks to be called with the value `1` before `0`. (I'm grateful they don't though!)

@fivdi thank you for the work you've put into `onoff`!

fivdi, 2022-05-15:
> I hope I can use this thread for a related question as I don't want to log an issue if there is no need.
> 
> I `watch` a few pins setup with `"in"` and `"both"`. I have a few PIR sensors wired to these. With movement the callback value parameter is `1` and once idle the value parameter is `0`.
> 
> Whenever I write `1` to any of the other output pins (with cheap relays hooked up) the `watch` callbacks of all the input pins are called with the value `0`. The relays control 10W LED lights (set up not to interfere with the PIR sensors).
> 
> I suspect this is circuit related, but even so I would expect the `watch` callbacks to be called with the value `1` before `0`. (I'm grateful they don't though!)
> 
> @fivdi thank you for the work you've put into `onoff`!

I would also expect the problem to be circuit related rather than being software related. One way to confirm that the problem is circuit related would be to replace the relays with LEDs. If everything works as expected when using LEDs rather than relays I'd say there's a problem with the circuit. If there is still a problem when LEDs are used, it may be a software problem.

Regarding your expectations of the value being `1` or `0` when the `watch` callback is called; if circuit problems with relays are causing very short spikes in input pins, the input pin may have gone high and then low in a very short time. In this case, the onoff module may have read the pins value as 0 rather than 1.

gwest7, 2022-05-15:
This makes sense. I'll replace the relays with LEDs sometime and confirm this. Cheap hardware can sometime be quite expensive. 😉 

Thank you for sharing your insights.