fivdi/onoff #183: Initialize GPIO without change pin status.

elfive, 2021-01-22:
I wanna init a gpio instance(pin direction set to 'out') without change it's pin status('low' or 'high'),

I tried read it's status before set it's direction to 'out'.
since my pin previous direction is 'out', so if I set to 'in' and read it, because of direction change, it will always call ```gpio.setDirection(direction);``` which will set the pin status to low.

```
    const reconfigureDirection =
      typeof options.reconfigureDirection === 'boolean' ?
        options.reconfigureDirection : true;

    const requestedDirection =
      direction === 'high' || direction === 'low' ? 'out' : direction;

    if (reconfigureDirection || gpio.direction() !== requestedDirection) {
      gpio.setDirection(direction);
    }
```

Is there a possiable way to accomplish my goal?

fivdi, 2021-01-22:
> I wanna init a gpio instance(pin direction set to 'out') without change it's pin status('low' or 'high'),

If I understand the requirement correctly, I would expect something like the following to work:
```js
const Gpio = require('onoff').Gpio;
const led = new Gpio(17, 'out', {reconfigureDirection: false});
...
```

fivdi, 2021-01-26:
@elfive was is possible to make progress here? Did the proposed solution work?

elfive, 2021-01-26:
> If I understand the requirement correctly, I would expect something like the following to work:
>
> ```js
> const Gpio = require('onoff').Gpio;
> const led = new Gpio(17, 'out', {reconfigureDirection: [...]

Sorry, the suggestion won’t work.
when direction set to "out", It will automatically set the pin status to LOW……

fivdi, 2021-01-26:
> when direction set to "out", It will automatically set the pin status to LOW……

This is not correct. When the `reconfigureDirection` option is set to `false`, the state of a pin that is already configured as an output (already exported) will not be modified.

What have you tried and what were the problems? Can you post a complete but short program that can be used to reproduce the problem please?

elfive, 2021-01-26:
I’m writing a homebridge plugin,you can find my code here: https://github.com/elfive/homebridge-raspberry-simplegpio

fivdi, 2021-01-26:
If you can post a complete but short program that can be used to reproduce the issue I can look into the problem. The short program shouldn't use homebridge. What I can't do is learn about homebridge to debug a problem that you are having.

closed 2021-01-26 by elfive