fivdi/onoff #184: GPIO not working in a worker_thread - nodejs 14.15

lucianobustos, 2021-02-06:
Whe I execute the following code on a nodejs server app. it fails with the following error after GPIO receive a trigger event.

> WORKER: Thread started without issue
> FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope

## thread.js code
const { parentPort, workerData, isMainThread } = require('worker_threads');
const Gpio = require('onoff').Gpio;
//we are in a worker thread.
  if (!isMainThread) {
      console.log('WORKER: Initializing');
      console.log(workerData);
      const gpio = new Gpio(4, 'in' 'rising');
      gpio.watch((err, _val) => {
         parentPort.postMessage(_val);
      });
      console.log('WORKER: Thread started without issue');
  }

## main thread.js
const {Worker} = require('worker_threads');
const path = require('path');
const wScript = path.join(__dirname, './thread.js');
const worker = new Worker(wScript, {workerData: {data:1, foo:'bar'});
worker.on('message', (data) => console.log(data));
worker.on('error', (e) => console.error(e));
worker.on('exit', () => console.log('Worker Exit'));

## Error

The error when i push a button on pin4 of raspberry. when GPIO detect a trigger.

Did someone experience this???

fivdi, 2021-02-06:
I'm afraid this is to be expected a the moment. onoff uses the [epoll module](https://github.com/fivdi/epoll) do detect hardware interrupts. epoll is a native Node.js add-on implemented in C++. Native add-ons can only be loaded from multiple threads if they fulfil [certain conditions](https://nodejs.org/dist/latest-v15.x/docs/api/addons.html#addons_worker_support). epoll currently doesn't fulfil these conditions.

lucianobustos, 2021-02-06:
Yes. Thanks for pointing that. I was considering to write my own c++ extension form zero to move my calculations one layer down and avoid the nodejs race condition .. but I'm glad I found your other library... 😀

closed 2021-02-06 by lucianobustos