fivdi/onoff #185: EACCES: permission denied, open '/sys/class/gpio/export' as superuser

sgergo, 2021-02-23:
Starting my nodejs app on a headless Pi as a superuser gives me the following. Starting it as non-root works well. It is rather strange, can you help me?

pi@raspberrypi:~/pw/as-cm4 $ sudo npm start

> as-config-portal@0.0.0 start
> node ./bin/www

node:internal/fs/utils:322
    throw err;
    ^

Error: EACCES: permission denied, open '/sys/class/gpio/export'
    at Object.openSync (node:fs:492:3)
    at Object.writeFileSync (node:fs:1516:35)
    at exportGpio (/home/pi/pw/as-cm4/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/pw/as-cm4/node_modules/onoff/onoff.js:172:36)
    at Object.<anonymous> (/home/pi/pw/as-cm4/pinhandling.js:3:18)
    at Module._compile (node:internal/modules/cjs/loader:1108:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1137:10)
    at Module.load (node:internal/modules/cjs/loader:973:32)
    at Function.Module._load (node:internal/modules/cjs/loader:813:14)
    at Module.require (node:internal/modules/cjs/loader:997:19) {
  errno: -13,
  syscall: 'open',
  code: 'EACCES',
  path: '/sys/class/gpio/export'
}

fivdi, 2021-02-28:
The way in which onoff is implemented should prevent onoff from being the cause of EACCES errors.

If Raspberry Pi OS (previously called raspbian) is being used, then user pi has full access to GPIOs without the need to use sudo. A Node.js application using onoff run by user pi should be able to access GPIOs without sudo.

I think the system where the issue is being seen is configured incorrectly and that this is resulting in EACCES errors.

Please install onoff in a newly created directory to see if your application then works from that directory. When installing and using onoff, don't use sudo at any point.

Please use the following commands to install onoff:

```bash
pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ mkdir onoff-test
pi@raspberrypi:~ $ cd onoff-test/
pi@raspberrypi:~/onoff-test $ npm list
npm WARN npm npm does not support Node.js v10.23.1
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
/home/pi/onoff-test
└── (empty)

pi@raspberrypi:~/onoff-test $ npm install onoff
```

One of the commands run above is `npm list`. To ensure that no erroneous installations of onoff are being used, please confirm that the output of `npm list` is:

```bash
/home/pi/onoff-test
└── (empty)
```

fivdi, 2021-03-14:
I'll go ahead and close this as it's very unlikely to be a problem with onoff. If it does turn out to be a problem with onoff the issue can be reopened.

closed 2021-03-14 by fivdi