fivdi/onoff #188: TypeError: writeSync is not a function

SubjectXXX, 2021-04-07:
Greetings,

This is probably not onoff bug, but it only happens when using onoff functionality, so I decided to ask here.

I have an array of objects:
objects[0] = {id: 0, type: "shade", group: "shades", name: "All", status: 2, full_timer: 20, moving_timer: 0, position: 1, active_timer_id : 0, start_timer: 0, pin_up: 18, pin_down: 18, commands: ["Dol", "Gor", "Stop"], states: ["up", "down", "stop"], auto_timer_enabled: 0, auto_timer: [0,"17:00"]};
objects[1] = {id: 1, type: "shade", group: "shades", name: "Kitchen", status: 2, full_timer: 10, moving_timer: 0, position: 1, active_timer_id : 0, start_timer: 0, pin_up: 0, pin_down: 1, commands: ["Dol", "Gor", "Stop"], states: ["up", "down", "stop"], auto_timer_enabled: 1, auto_timer: [0, "20:15"]};

Some where in code I need to initialize new Gpio object on selected pin with 'out':
if (typeof objects[item_id].gpio_up === 'undefined') {
	objects[item_id].gpio_up = new Gpio(7, 'out');
}

But then, if I do this:
objects[item_id].gpio_down.writeSync(1);

I get
TypeError: objects[item_id].gpio_up.writeSync is not a function

Problem is, it works, if I make new Gpio object on the fly and do operation on it:
var gpio_item = new Gpio(7, 'out');
gpio_item.writeSync(1);

but I need to have access to this objects for later use (optimization on rpi4 with latest nodejs and onoff), since 
var gpio_item = new Gpio(7, 'out');
always puts pin to 0v.

I need to read voltage on pin and then decide what to do later in code.

Where am I doing the wrong turn?

fivdi, 2021-04-07:
> This is probably not onoff bug, but it only happens when using onoff functionality, so I decided to ask here.

Yes, I don't think this is an onoff bug either.

> Some where in code I need to initialize new Gpio object on selected pin with 'out':
> if (typeof objects[item_id].gpio_up === 'undefined') {
> objects[item_id].gpio_up = new Gpio(7, 'out');
> }
>
> But then, [...]

I think there is a typo in the example code here and `objects[item_id].gpio_down.writeSync(1)` should be `objects[item_id].gpio_up.writeSync(1)`. Nevertheless, even if `objects[item_id].gpio_up.writeSync(1)` is being used and the `TypeError` is thrown, it most likely means one of two things:

 * Somewhere in your code, a value that is not a Gpio object is being assigned to `objects[item_id].gpio_up`
 * Somewhere in your code, a value that is not a Function is being assigned to `objects[item_id].gpio_up.writeSync`

> Problem is, it works, if I make new Gpio object on the fly and do operation on it:
> var gpio_item = new Gpio(7, 'out');
> gpio_item.writeSync(1);

Yes, this works because neither `gpio_item` nor `gpio_item.writeSync` is being overwritten.

> but I need to have access to this objects for later use (optimization on rpi4 with latest nodejs and onoff), since
> var gpio_item = new Gpio(7, 'out');
> always puts pin to 0v.

Please carefully read the documentation for the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) to discover the different possibilities here.

> I need to read voltage on pin and then decide what to do later in code.

For this `Gpio(<number>, 'in')` is required. See the documentation for further information.

> Where am I doing the wrong turn?

I can't say as I don't know your code. You will need to debug the application to see what is going wrong.

SubjectXXX, 2021-04-08:
Thank you for your quick reply.

Sorry for edit, I did edit it in first minute after posting tho :)

Anyway, regarding changing object or function somewhere else in code  is not a problem, since I only make this object once and at the same time I save it to array.
Later, I only access it via that function call.. Unless there are same backstage shenanigans NodeJS does..

I have read constructor instructions and didnt find where I can instantiate GPIO object while retaining already said value on it. Thus I went direction of saving created objects into array.

By "I need to read voltage.." I meant I need to know, what is the current state of pin (1 or 0) in out mode and then act accordingly (sounds really complicated, but in my case, I need to check if shades on my window is moving (pin is active, 1) and then stop it before moving shade in different direction).

I did workaround that I now strictly always set pins to 0 no matter what, then stop moving shade in selected direction. But its ugly code..

fivdi, 2021-04-08:
> Anyway, regarding changing object or function somewhere else in code is not a problem, since I only make this object once and at the same time I save it to array.
Later, I only access it via that function call.. Unless there are same backstage shenanigans NodeJS does..

Node doesn't do any backstage shenanigans 😄. Please post a short, complete and tested program that can be used to reproduce the `TypeError` problem. 

> I have read constructor instructions and didn't find where I can instantiate GPIO object while retaining already set value on it. Thus I went direction of saving created objects into array.

When a Raspberry Pi is powered on all GPIOs are configured as inputs by default. Because they are inputs, they are not outputs, so it's not possible to configure them as outputs which retain their current output value. They don't have an output value that could be retained when powered on, they have an input value.

If the GPIO has already been configured as an output, for example, by a previous run of your application (after the Raspberry Pi was powered on), the `reconfigureDirection` option described in the constructor can be set to `false` to leave the GPIO in it's current state.

SubjectXXX, 2021-04-08:
Ahh, you pushed me into the right direction.

Problem was that I was saving this array to DB for persistent settings after possible power cutout, and I was saving this gpio objects into same database, when application restarted, values were read from DB (shades position and stuff) and also an object, which wasnt gpio anymore, but some placeholder with default methods and properties.

And also, yes, I missed reconfigureDirection (direction was a bit confusing word for me in this context thus I skipped the whole part), I will fix constructors with said optional switch and will always generate new gpio objects, not gonna save them for later use.

Thank you for your help.

closed 2021-04-08 by SubjectXXX