fivdi/onoff #189: Raspberry Pi GPIO control in electron App

lakshanthad, 2021-06-23:
I have created a simple app that can control the GPIO pins on the Raspberry Pi with a simple button press. However, when I press the button, there is no response from the GPIO pins (here connected to GPIO 24). Also, I have already installed the onoff module from `npm install onoff`. When I create a simple javascript file with only GPIO control and run `node <file_name>.js`, it runs successfully. But when I try to integrate into electron, it stops working. Please check the following code. Help would be appreciated.

<h4>index.html</h4>

```html
<!DOCTYPE html>
<html>
    <body>
        <h1>GPIO Control</h1>
        <button id="Btn" class="button is-primary">LED ON</button>
        <script src="./render.js"></script>
    </body>
</html>
```

<h4>main.js</h4>

```javascript
const { app, BrowserWindow } = require('electron')
const path = require('path')

function createWindow () {
  const mainWindow = new BrowserWindow({
    width: 800,
    height: 600,
    webPreferences: {
        nodeIntegration: true
      }
  })

  mainWindow.loadFile('index.html')
}

app.whenReady().then(() => {
  createWindow()

  app.on('activate', function () {
    if (BrowserWindow.getAllWindows().length === 0) createWindow()
  })
})

app.on('window-all-closed', function () {
  if (process.platform !== 'darwin') app.quit()
})
```

<h4>render.js</h4>

```javascript
var Gpio = require('onoff').Gpio; 
var LED = new Gpio(24, 'out');

const Btn = document.getElementById("Btn");
Btn.onclick = function() {
  myFunction()
};

function myFunction() {
  LED.writeSync(1);
}
```

fivdi, 2021-06-23:
I'm afraid I'm not familiar with Electron and can't provide much help. Are error messages displayed when the application is run?

fivdi, 2021-07-01:
Duplicate of https://github.com/atom/atom/discussions/22638

onoff depends on a native Node module called [epoll](https://github.com/fivdi/epoll/). Special steps need to be taken when installing native Node modules like epoll with Electron. For additional information see [here](https://www.electronjs.org/docs/tutorial/using-native-node-modules).

If these special steps are not taken into account, errors complaining about incorrect `NODE_MODULE_VERSION` numbers like [the errors shown here](https://github.com/fivdi/pigpio/issues/133) can occur.

@lakshanthad Is your application showing `NODE_MODULE_VERSION` errors or other errors?

leerlemoniii, 2021-07-01:
might be the missing semi colon on myFunction call

Btn.onclick = function() {
  myFunction()
};
also do not see in this code where you are loading the gpio library

lakshanthad, 2021-07-02:
Hello. Thank you all for the replies. I have now made it work using ipcMain and ipcRenderer modules. Not sure whether there is an easy way to do this

fivdi, 2021-07-02:
Thank you for the feedback. I guess if it's working now the issue can be closed.

closed 2021-07-02 by fivdi

NathanJ4620, 2021-08-30:
the ipcMain and ipcRenderer are the only ways to communicate between electron and an external Node instance. This functionality is to help maintain security on the host device when creating a shipped application.

TheGrateMartian, 2021-10-17:
I am trying to do the same. @lakshanthad  Can you share please show/share what you did to get it workin?

HelloSmartFactory, 2022-01-24:
@TheGrateMartian 
Here is a sample using ipc.
https://github.com/boraciner/electron-gpio