fivdi/onoff #192: Raspberry 4 - malloc(): corrupted top size

dariocavada, 2021-12-17:
If I try to use it on raspberry 4, only by including the library I get this error:

code to reproduce the error:
``` js
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
console.log('onoff require test');
```

```
malloc (): corrupted top size
Aborted
```

Tried with both nodejs 14, 16 and 17

Any suggestions or work arounds ?

closed 2021-12-17 by dariocavada

developer-ishan, 2022-07-05:
What was the work around

mesutrk95, 2022-10-13:
have you got any fix for this problem?

fivdi, 2022-10-15:
I can't reproduce this issue with Node.js v18.8.0 and onoff v6.0.3 on a Raspberry Pi 4 running Raspberry Pi OS. Here is what I see when running the program provided in the initial post above. 

```
pi@raspberrypi:~/test $ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

pi@raspberrypi:~/test $ node --version
v18.8.0

pi@raspberrypi:~/test $ npm list --all
test@ /home/pi/test
└─┬ onoff@6.0.3
  ├─┬ epoll@4.0.1
  │ ├─┬ bindings@1.5.0
  │ │ └── file-uri-to-path@1.0.0
  │ └── nan@2.17.0
  └── lodash.debounce@4.0.8

pi@raspberrypi:~/test $ cat test.js 
var Gpio = require('onoff').Gpio; //include onoff to interact with the GPIO
console.log('onoff require test');

pi@raspberrypi:~/test $ node test.js 
onoff require test

pi@raspberrypi:~/test $ 
```

As can be seen, there is no `malloc (): corrupted top size` error. The program successfully requires onoff and displays the expected message.

Please provide the exact details required for reproducing the issue.