fivdi/onoff #197: Pi 5 Compatibility issues

rhysmorgan134, 2023-11-06:
Hi,

First huge thanks to all your libraries, both this and spi-device have been amazing for me!

I have a pi5, and I am trying to get some of my packages running on it, however it seems that this library (and most likely spi-device) are not working. Unfortunately I really don't have much in the way of error logs, as as far out logging the library seems to be working, however, when I set an interrupt on a pin, and then drive that pin, the onoff library doesn't seem to pick up on a value change. When running poll on pinctrl via command line though, I can clearly see that values changing.

I suspect this may be quite a large issue due to the rpi5 now using the rp1 chip for gpio, and from the small amount of information I can find, it seems this is not backwards compatible with certain implementations

fivdi, 2023-11-09:
onoff uses the GPIO sysfs interface for userspace GPIO.  This interface has been deprecated for quite some time (see [here](https://www.kernel.org/doc/Documentation/gpio/sysfs.rst)). I don't know if the GPIO sysfs Interface for userspace GPIO is supported on the Raspberry Pi 5 so I asked [here](https://forums.raspberrypi.com/viewtopic.php?t=359302) on the Raspberry Pi forum.

fivdi, 2023-11-09:
@rhysmorgan134 Please take a look at the [forum post](https://forums.raspberrypi.com/viewtopic.php?t=359302). I need to do some more testing but everything appears to be working, including interrupt detection. However, the GPIO numbering is completely different.

rhysmorgan134, 2023-11-10:
Thanks alot @fivdi just ran some tests now, and after upgrading I appear to now be getting these errors, I will have an investigate!

```
node:internal/fs/utils:350
    throw err;
    ^

Error: EINVAL: invalid argument, write
    at Object.writeSync (node:fs:936:3)
    at Object.writeFileSync (node:fs:2330:26)
    at exportGpio (/home/pi/SocketMost/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/SocketMost/node_modules/onoff/onoff.js:172:36)
    at new OS8104A (/home/pi/SocketMost/dist/driver/OS8104A.js:46:26)
    at new SocketMost (/home/pi/SocketMost/dist/server/SocketMost.js:107:23)
    at Object.<anonymous> (/home/pi/SocketMost/examples/server.js:3:20)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32) {
  errno: -22,
  syscall: 'write',
  code: 'EINVAL'
}

Node.js v18.18.2
```

rhysmorgan134, 2023-11-10:
Never mind, I thought the new gpio numbers were now the pin numbers, but they are the 4xx range!

rhysmorgan134, 2023-11-10:
As an extra spi-device works out of the box, great stuff, and thanks for the help and the libraries!

closed 2023-11-10 by rhysmorgan134

fivdi, 2023-11-11:
@rhysmorgan134 Thanks for confirming that both onoff and spi-device are now working for you.

rodsnts, 2024-02-15:
> Never mind, I thought the new gpio numbers were now the pin numbers, but they are the 4xx range!

Hi @rhysmorgan134 , what you did to be able to use onoff with the raspberry pi 5?

I have this simple code:
```
const Gpio =  require("onoff").Gpio;
const led = new Gpio(4, "out");

led.writeSync(1);
```

And I get a error message saying:
`Error: EINVAL: invalid argument, write`

fivdi, 2024-02-16:
@rodsnts To understand how GPIO numbering works on the Pi5, it's important to read this post: https://forums.raspberrypi.com/viewtopic.php?t=359302

Assuming you want to use GPIO4, which is PIN7 on the GPIO header, then you'll need to replace this:
```javascript
const Gpio =  require("onoff").Gpio;
const led = new Gpio(4, "out");

led.writeSync(1);
```
with this:
```javascript
const Gpio =  require("onoff").Gpio;
const led = new Gpio(403, "out");

led.writeSync(1);
```

In other words, what used to be GPIO4 is now GPIO403 on the Pi5. Give this a try to see if it works.

rodsnts, 2024-02-19:
@rodsnts Worked, please marry me.

rodsnts, 2024-02-19:
@fivdi Worked!! Please marry me.