fivdi/onoff #199: RPI 5 gpio numbers

livep2000, 2024-04-13:
Recently I burned out my rpi4, just at the moment my highly complicated radio transceiver reached some point of functionality. 
Ordered a pi5, to discover its not working. After reading and reading this issues tracker give the fairly simple solution.

Just add 571 to the gpio pin number. Proposing something like this to make it fully compatible:

```
let file = fs.readFileSync("/proc/cpuinfo", "utf8");
let arr = file.split(/\r?\n/);
arr.forEach((line, idx)=> {
    if(line.includes("Pi 5")){
        console.log(line);
        gpio += 571;
        }
});
```
Imre

SamDecrock, 2024-05-24:
I don't think it's Pi5 related but rather the version of Raspbian. On Bullseye it works, on Bookworm it's broken.

nwpray, 2024-06-27:
I opened a PR to lookup the base address and add that automatically to the address you pass
https://github.com/fivdi/onoff/pull/204

gdorbes, 2025-07-22:
The gpio number issue is a matter of bath RPi model and os version. My tests:

- Bookworm on RPI5 add 571
- Bookworm on RPi4B add 522

To find the right value, see `cat /sys/kernel/debug/gpio`

fq-selbach, 2026-02-10:
> I agree with 95% of your comment. It is the reason why I have developped a library based on libgpiod for recent RPi hardware and OS. See https://github.com/gdorbes/rpi-io 😄

I'm looking at it right now and will try to use it as an 'onoff' replacement :-)

> For the remaining 5%, I must admit that I still use sysf interface to manage hardware-based PWM interface. I did not find a "modern" solution to replace sysf.

As far as I understand it the PWM part of sysfs (`/sys/class/pwm/`) is actually NOT deprecated, only the GPIO part (`/sys/class/gpio`). Its a bit confusing, so I'm not 100% certain -_-

OlliMartin, 2026-04-26:
As far as I can tell this is not related to the library because you can change the offset, but for anyone stumbling over this as well: 
I cleanly installed a raspberry pi5 16gb with the official raspberry image 2-3 days back - offset seems to (sometimes?) be 569, not 571.

```
$ cat /sys/kernel/debug/gpio

// [..]
gpiochip0: GPIOs 569-622, parent: platform/1f000d0000.gpio, pinctrl-rp1:
 gpio-569 (ID_SDA              )
 gpio-570 (ID_SCL              )
```