fivdi/onoff #200: fs error

sk337, 2024-04-30:
```
node:internal/fs/utils:347
    throw err;
    ^

Error: EINVAL: invalid argument, write
    at Object.writeSync (node:fs:920:3)
    at Object.writeFileSync (node:fs:2208:26)
    at exportGpio (/home/pi/rpiapp/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/rpiapp/node_modules/onoff/onoff.js:172:36)
    at new s (/home/pi/rpiapp/dist/index.cjs:1:1479)
    at Object.<anonymous> (/home/pi/rpiapp/dist/index.cjs:1:2592)
    at Module._compile (node:internal/modules/cjs/loader:1198:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1252:10)
    at Module.load (node:internal/modules/cjs/loader:1076:32)
    at Function.Module._load (node:internal/modules/cjs/loader:911:12) {
  errno: -22,
  syscall: 'write',
  code: 'EINVAL'
}
pi@raspberrypi:~/rpiapp $ node -v
v16.20.2
```

michaelpeterlee, 2024-04-30:
Same here with RPi5 Node v18.19.0

michaelpeterlee, 2024-04-30:
Workaround: https://github.com/fivdi/onoff/issues/199

alex-reach, 2024-05-01:
It seems that Raspberry OS kernel addresses GPIO pins yet by another numbering scheme.

cat /sys/kernel/debug/gpio
`gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               )
 gpio-520 (GPIO8               )
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              )
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
`
You must use the number after gpio-[number]

szwacz, 2024-05-14:
Looks like every GPIO library when run on new kernel will have this issue. At the same time seems like the whole community is perplexed what actually happened. Did you manage to find any official explanation how to prepare for this change and why it was introduced?

SamDecrock, 2024-05-24:
On Bullseye it works, on Bookworm it's broken.

nwpray, 2024-06-27:
The library needs to look for the `gpiochip*` file and add the `*` value to every pin. on mine its `gpiochip512` so for pin 2 i believe it would be `514`

nwpray, 2024-06-27:
I opened a PR to lookup the base address and add that automatically to the address you pass
https://github.com/fivdi/onoff/pull/204

sberkovitz, 2024-06-27:
Awesome @nwpray  - though it seems onoff hasn't been touched in over 3 years, wondering if we should just fork it and modify it for our own use at this point?